huggingface/lerobot #141: Visualization during training

fabawi, 2024-05-06:
Thank you for this great framework. I tried to train a model following the example:

```python lerobot/scripts/train.py     policy=act     env=aloha     env.task=AlohaInsertion-v0     dataset_repo_id=lerobot/aloha_sim_insertion_human```

But it seems the `render_mode` is hardcoded as `rgb_array` in `lerobot/common/envs/factory.py' and the environment does not support `human`. I was wondering if there are plans to support non-headless training. It would be very useful for debugging the models, more so if it could somehow be linked to the rerun GUI. 

Are there plans to support that? If not, do you have any tips for using the rerun dataset visualizer, or perhaps the [dm_control viewer](https://github.com/google-deepmind/dm_control/blob/main/dm_control/viewer/README.md) for visualizing the aloha environment during training/evaluation?

Cadene, 2024-05-06:
I would recommend using wandb (which is set by default if you do `wandb login`). As you can see on the top left, we upload videos of the evaluation during training:
[image]

See the training section of the README for more info: https://github.com/huggingface/lerobot?tab=readme-ov-file#train-your-own-policy

We actually run evaluation on multiple parallel environments and only send the rendering of the first one to wandb:
https://github.com/huggingface/lerobot/blob/main/lerobot/scripts/train.py#L353

Note that we render 4 episodes (see: https://github.com/huggingface/lerobot/blob/main/lerobot/scripts/train.py#L348) that you can visualize as mp4 in your experiment directory.

"I was wondering if there are plans to support non-headless training."
> We don't plan to do so. We tend to prefer headless training as we directly work on a remote cluster.
If you have a specific use case in mind that requires non-headless, let us know.

Cadene, 2024-05-06:
Also take a look at this example where it would be easier to add non-headless visualization by setting `render_mode="human"`:
https://github.com/huggingface/lerobot/blob/main/examples/2_evaluate_pretrained_policy.py#L35-L39

fabawi, 2024-05-06:
Thank you, I didn't realize that the videos were uploaded to wandb (it didn't upload any videos in my case, but probably something wrong with my setup). I can't think of a pressing reason for non-headless training.  But down the line, an online training paradigm with human feedback might require such a mechanism. The wandb video upload should be sufficient for now

alexander-soare, 2024-05-07:
@fabawi also, the training outputs folder should contain videos 
[image]

alexander-soare, 2024-05-07:
Closed, as it looks like you're satisfied with the outcome for now :)

closed 2024-05-07 by alexander-soare