huggingface/lerobot #142: The LeRobot TODO hyperlink in CONTRIBUTING.md is not working

vinayakkgarg, 2024-05-06:
### System Info

```Shell
The "LeRobot TODO" hyperlink in the contribution guide (lerobot/CONTRIBUTING.md) redirects to a 404 page.
The hyperlink currently redirects to: https://github.com/orgs/huggingface/projects/46
```

### Information

- [ ] One of the scripts in the examples/ folder of LeRobot
- [ ] My own task or dataset (give details below)

### Reproduction

1. Click the "LeRobot TODO" hyperlink in the contribution guide (lerobot/CONTRIBUTING.md)

### Expected behavior

The hyperlink should ideally redirect to a page with a list of to-do tasks.

alexander-soare, 2024-05-07:
Thanks @vinayakkgarg for reporting this. We forgot to make the board public. We'll get it organized and update the link shortly. cc @Cadene @aliberts

Cadene, 2024-05-10:
Fixed ;) https://github.com/orgs/huggingface/projects/46

closed 2024-05-10 by Cadene