huggingface/lerobot #143: running `python lerobot/scripts/eval.py \     -p lerobot/diffusion_pusht \     eval.n_episodes=10 \     eval.batch_size=10` on apple silicon Mac

ramkumarkoppu, 2024-05-07:
### System Info

```Shell
- `lerobot` version: 0.1.0
- Platform: macOS-14.4.1-arm64-arm-64bit
- Python version: 3.10.14
- Huggingface_hub version: 0.21.4
- Numpy version: 1.26.4
- PyTorch version (GPU?): 2.3.0 (False)
- Cuda version: N/A
- Using GPU in script?: not sure, running the unmodified out of the box script
- Using distributed or parallel set-up in script?: not sure
```

### Information

- [X] One of the scripts in the examples/ folder of LeRobot
- [ ] My own task or dataset (give details below)

### Reproduction

run `python lerobot/scripts/eval.py \     -p lerobot/diffusion_pusht \     eval.n_episodes=10 \     eval.batch_size=10` on apple silicon Mac gives following error at the end:

(lerobot) Rams-MBP:lerobot ramkumarkoppu$ python lerobot/scripts/eval.py \
>     -p lerobot/diffusion_pusht \
>     eval.n_episodes=10 \
>     eval.batch_size=10
README.md: 100%|████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████| 2.56k/2.56k [00:00<00:00, 20.2MB/s]
eval_avg_max_reward.csv: 100%|██████████████████████████████████████████████████████████████████████████████████████████████████████████████| 537/537 [00:00<00:00, 6.24MB/s]
eval_pc_success.csv: 100%|██████████████████████████████████████████████████████████████████████████████████████████████████████████████████| 236/236 [00:00<00:00, 2.15MB/s]
.gitattributes: 100%|███████████████████████████████████████████████████████████████████████████████████████████████████████████████████| 1.52k/1.52k [00:00<00:00, 14.6MB/s]
config.json: 100%|██████████████████████████████████████████████████████████████████████████████████████████████████████████████████████| 1.01k/1.01k [00:00<00:00, 11.3MB/s]
config.yaml: 100%|██████████████████████████████████████████████████████████████████████████████████████████████████████████████████████| 2.36k/2.36k [00:00<00:00, 29.1MB/s]
demo.gif: 100%|███████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████| 601k/601k [00:00<00:00, 5.45MB/s]
train_loss.csv: 100%|███████████████████████████████████████████████████████████████████████████████████████████████████████████████████| 20.7k/20.7k [00:00<00:00, 59.2MB/s]
training_curves.png: 100%|██████████████████████████████████████████████████████████████████████████████████████████████████████████████| 28.6k/28.6k [00:00<00:00, 4.40MB/s]
eval_info.json: 100%|████████████████████████████████████████████████████████████████████████████████████████████████████████████████████| 83.9k/83.9k [00:00<00:00, 896kB/s]
model.safetensors: 100%|████████████████████████████████████████████████████████████████████████████████████████████████████████████████| 1.05G/1.05G [00:59<00:00, 17.8MB/s]
Fetching 11 files: 100%|█████████████████████████████████████████████████████████████████████████████████████████████████████████████████████| 11/11 [00:59<00:00,  5.44s/it]
Traceback (most recent call last):█████████████████████████████████████████████████████████████████████████████████████████████████████▊| 1.05G/1.05G [00:58<00:00, 18.2MB/s]
  File "/Users/ramkumarkoppu/GIT/lerobot/lerobot/scripts/eval.py", line 604, in <module>
    eval(pretrained_policy_path=pretrained_policy_path, config_overrides=args.overrides)
  File "/Users/ramkumarkoppu/GIT/lerobot/lerobot/scripts/eval.py", line 508, in eval
    get_safe_torch_device(hydra_cfg.device, log=True)
  **File "/Users/ramkumarkoppu/miniconda3/envs/lerobot/lib/python3.10/site-packages/lerobot/common/utils/utils.py", line 17, in get_safe_torch_device
    assert torch.cuda.is_available()
AssertionError**

### Expected behavior

For apple silicon Mac, script should not expect cuda instead it can use mps if it can

qgallouedec, 2024-05-07:
Thanks for reporting. Please refer to https://github.com/huggingface/lerobot/issues/105#issuecomment-2096020509

aliberts, 2024-06-14:
Using `device=cpu` should work.
As for `mps` we currently don't support this backend, though we might in the future when its easier to support.

closed 2024-06-14 by aliberts