huggingface/lerobot #156: AttributeError: module 'rerun' has no attribute 'init' when running the "Visualize datasets" line specified in README

oschan77, 2024-05-09:
### System Info

```Shell
- `lerobot` version: 0.1.0
- Platform: Linux-6.5.0-28-generic-x86_64-with-glibc2.35
- Python version: 3.10.14
- Huggingface_hub version: 0.23.0
- Numpy version: 1.23.1
- PyTorch version (GPU?): 2.3.0+cu121 (True)
- Cuda version: 12010
- Using GPU in script?: No
- Using distributed or parallel set-up in script?: No
```

### Information

- [X] One of the scripts in the examples/ folder of LeRobot
- [ ] My own task or dataset (give details below)

### Reproduction

1. Run the "Visualize datasets" line specified in README.
```
python lerobot/scripts/visualize_dataset.py \
    --repo-id lerobot/pusht \
    --episode-index 0
```

It gives the following error:
```
Fetching 222 files: 100%|██████████████████████████████████████████████████████████████████████████████████| 222/222 [00:00<00:00, 377589.41it/s]
Traceback (most recent call last):
  File "/home/ubuntu/lerobot/lerobot/scripts/visualize_dataset.py", line 245, in <module>
    main()
  File "/home/ubuntu/lerobot/lerobot/scripts/visualize_dataset.py", line 241, in main
    visualize_dataset(**vars(args))
  File "/home/ubuntu/lerobot/lerobot/scripts/visualize_dataset.py", line 117, in visualize_dataset
    rr.init(f"{repo_id}/episode_{episode_index}", spawn=spawn_local_viewer)
AttributeError: module 'rerun' has no attribute 'init'
```

### Expected behavior

According to README, it will open rerun.io and display the camera streams, robot states and actions.

aliberts, 2024-05-09:
That's weird.
Can you paste here the result of `pip freeze | grep rerun-sdk` or `poetry show rerun-sdk` to see what version you have?

oschan77, 2024-05-10:
> That's weird. Can you paste here the result of `pip freeze | grep rerun-sdk` or `poetry show rerun-sdk` to see what version you have?

@aliberts I have rerun-sdk==0.15.1.

pip freeze | grep rerun-sdk
```
rerun-sdk==0.15.1
```

aliberts, 2024-05-10:
A few questions to try and narrow it down:
- Did you install lerobot from a fresh virtual environment with nothing inside?
- Any chance that you have [rerun](https://pypi.org/project/rerun/) installed in the venv you are using that would confuse python to import it instead of [rerun-sdk](https://pypi.org/project/rerun/) when doing `import rerun as rr`?
- Can you paste here what is returned by `python -c "import rerun; print(rerun.__file__)"`?
- Can you try to downgrade rerun-sdk at something like 0.14.1?

oschan77, 2024-05-13:
@aliberts Thank you so much for your help! Somehow I had another package called "Rerun" installed with "rerun-sdk". Once I uninstalled "Rerun", everything works as expected. 

In my case, rerun-sdk==0.15.1 does not account for this issue.

closed 2024-05-13 by oschan77

milong26, 2025-11-10:
I met this problem too. I do like this:
```shell
pip uninstall rerun
pip uninstall rerun-sdk
pip install -e ".[feetech]" (install what u need)
```
It said  
```
Installing collected packages: rerun-sdk, lerobot
  Attempting uninstall: lerobot
    Found existing installation: lerobot 0.4.1
    Uninstalling lerobot-0.4.1:
      Successfully uninstalled lerobot-0.4.1
Successfully installed lerobot-0.4.1 rerun-sdk-0.26.2
```