huggingface/lerobot #167: python3.10 how to install rerun-sdk

renhehehe, 2024-05-10:
### System Info

```Shell
ubuntu18.04
python3.10

ERROR: Could not find a version that satisfies the requirement rerun-sdk>=0.15.1 (from lerobot) (from versions: none)
ERROR: No matching distribution found for rerun-sdk>=0.15.1
```

### Information

- [X] One of the scripts in the examples/ folder of LeRobot
- [ ] My own task or dataset (give details below)

### Reproduction

pip install .

ERROR: Could not find a version that satisfies the requirement rerun-sdk>=0.15.1 (from lerobot) (from versions: none)
ERROR: No matching distribution found for rerun-sdk>=0.15.1

### Expected behavior

I want to know how to solve this problem

aliberts, 2024-05-10:
Hi, thanks for reporting this.

From [this issue](https://github.com/rerun-io/rerun/issues/1320#issuecomment-1441481186), it looks like there's a problem with glibc on Ubuntu 18.04
Can you try to build it from source? I'll try to replicate this in a container.

You might also find help on the aforementioned issue ;)

renhehehe, 2024-05-13:
thank you！ Ubuntu20.04 solved this problem

closed 2024-05-13 by renhehehe