huggingface/lerobot #168: aloha_static_thread_velcro videos are actually aloha_static_tape videos

andrew1ee, 2024-05-10:
Thank you for the great work! 

I was looking through the dataset and it seems that videos in `aloha_static_thread_velcro` dataset are actually `aloha_static_tape` videos. 

Also, most of the dataset which is supposed to be 50 episodes (checked the google drive) contain only 49 episodes (missing index 49).

Cadene, 2024-05-10:
@andrew1ee Indeed. It's a bug caused by `gdown` I am currently working on a fix. I uploaded all raw datasets on my page: https://huggingface.co/cadene
 I will open a PR soon.

Cadene, 2024-05-17:
@andrewcwlee Sorry for the late reply. PR is almost ready: https://github.com/huggingface/lerobot/pull/175

I visualized the new version of the dataset and `aloha_static_thread_velcro` corresponds to velcro, `aloha_static_tape` corresponds to tape
https://github.com/huggingface/lerobot/assets/4681518/266b5ed1-7d11-4495-b01f-581fc1907abc

Cadene, 2024-05-20:
Hey @andrewcwlee , the PR is merged. I made sure to fix this issue. Could you check on our side if the issue is solved? Dont hesitate to message me if you need any help.
Best