huggingface/lerobot #170: "A question about virtual environments"

le-wei, 2024-05-10:
### System Info

```Shell
python 3.10
```

### Information

- [ ] One of the scripts in the examples/ folder of LeRobot
- [X] My own task or dataset (give details below)

### Reproduction

Thank you very much for the excellent work you have done, which has facilitated my learning. I have a question. If I collect data using a real robot and am unable to create an env, what should I do? I have already converted my collected data into the Lerobot format, but during training and testing, I need to make_env an env for testing. How can I resolve this? Does this mean that this library can currently only be used in situations where there is already a virtual environment? Thank you very much for your response and advice.
I'm sorry I couldn't find a topic related to the inquiry, so I had to ask in the form of a bug report.

### Expected behavior

"I want to use this library to train on real data that I collected myself."

Cadene, 2024-05-10:
Hello @le-wei ,

Thanks for your kind words!

Indeed, we currently don't support "real-world" env, but we are working on it!

As you can see, we recently assembled our robot.
https://github.com/huggingface/lerobot/assets/4681518/8aeba2c5-3f1d-4f49-8002-9787d279abc8
[image]

We will open a PR next week to be able to connect LeRobot to a real-world hardware.

We didnt started yet, so if you have some code for a "gym-real-world" environment, dont hesitate to open a PR and we would be happy to iterate with you. You could maybe add this env in https://github.com/huggingface/lerobot/tree/main/lerobot/common/envs

Stay tuned ;)

Best

aliberts, 2024-07-23:
Closing this since #288 is now merged (we're continuing to add real-world support for more robots). Feel free to reopen or create another issue if need be ;)

closed 2024-07-23 by aliberts