huggingface/lerobot #178: Can not load a dataset from a Hugging Face repository

Joeland4, 2024-05-12:
### System Info

```Shell
lerobot
```

### Information

- [X] One of the scripts in the examples/ folder of LeRobot
- [ ] My own task or dataset (give details below)

### Reproduction

can not load datasets from HF when doing: python lerobot/scripts/visualize_dataset.py     --repo-id lerobot/pusht     --episode-index 0, or python -u example/1_load_lerobot_dataset.py
error: dataset = LeRobotDataset(repo_id)
ConnectionError: Couldn't reach 'lerobot/pusht' on the Hub (ConnectionError)

### Expected behavior

help me download the datasets from HF

aliberts, 2024-05-12:
Hi, thanks for reporting.
I'm unable to reproduce, are you using a proxy?

Also, can you try if this minimal code works?
```python
from datasets import load_dataset
from lerobot.common.datasets import video_utils

dataset = load_dataset("lerobot/pusht", split="train", download_mode='force_redownload')
```

Joeland4, 2024-05-12:
> Hi, thanks for reporting. I'm unable to reproduce, are you using a proxy?
>
> Also, can you try if this minimal code works?
>
> ```python
> from datasets import load_dataset
> from lerobot.common.datasets [...]

Thank you for your reply. Yes I am using a proxy, i get the List of available datasets ... but get the ConnectionError: Couldn't reach 'lerobot/pusht' on the Hub (ConnectTimeout). Also the minimal code doesn't work , same connection error.

Joeland4, 2024-05-12:
> Hi, thanks for reporting. I'm unable to reproduce, are you using a proxy?
>
> Also, can you try if this minimal code works?
>
> ```python
> from datasets import load_dataset
> from lerobot.common.datasets [...]

What should I modify to reproduce if I manually download the PushT dataset from HF URL?

aliberts, 2024-05-12:
Okay, maybe try that (fill in you actual proxies first):
```python
from datasets import DownloadConfig, load_dataset
from lerobot.common.datasets import video_utils

download_config = DownloadConfig(proxies={"https": "<your proxies>"}, force_download=True)
dataset = load_dataset("lerobot/pusht", split="train", download_config=download_config)
```

Joeland4, 2024-05-12:
> Okay, maybe try that (fill in your actual proxies first):
> 
> ```python
> from datasets import DownloadConfig, load_dataset
> from lerobot.common.datasets import video_utils
> 
> download_config = DownloadConfig(proxies={"https": "<your proxies>"}, force_download=True)
> dataset = load_dataset("lerobot/pusht", split="train", download_config=download_config)
> ```

Thanks god, I finally make it work by using hf-mirror. Thank you very much for your reply!

aliberts, 2024-05-12:
Awesome!

closed 2024-05-12 by aliberts

jackylee1, 2024-05-21:
> Thanks god, I finally make it work by using hf-mirror. Thank you very much for your reply!

hi please can you share how to change the code .