huggingface/lerobot #183: Diffusion policy can't be applied to other tasks？

Joeland4, 2024-05-14:
### System Info

```Shell
lerobot
```

### Information

- [X] One of the scripts in the examples/ folder of LeRobot
- [ ] My own task or dataset (give details below)

### Reproduction

python lerobot/scripts/train.py     policy=diffusion     env=aloha     env.task=AlohaInsertion-v0     dataset_repo_id=lerobot/aloha_sim_insertion_human

### Expected behavior

KeyError: 'observation.image'. I want to apply the diffusion policy to other tasks except 'pusht', but I found that the key in diffusion config '[observation.image]'  doesn't match the file name (some is observation.images.top) of the video datasets. So how can I make it available for other tasks?

aliberts, 2024-05-14:
Hey there,
We are working on it: #149 ;)
In any case, [`available_policies_per_env`](https://github.com/huggingface/lerobot/blob/89c6be84caf5a628c04adcba05821d70b054f25a/lerobot/__init__.py#L94) should serve as the reference for what's working with what.

Cadene, 2024-05-14:
Just for context, we plan to support diffusion policy for navigation tasks ;)
[See Nomad, best paper award finalist ICRA 2014](https://general-navigation-models.github.io/nomad)
If someone has time to help port the datasets and pretrained policies, that would be super useful!

Joeland4, 2024-05-15:
> Hey there, We are working on it: #149 ;) In any case, [`available_policies_per_env`](https://github.com/huggingface/lerobot/blob/89c6be84caf5a628c04adcba05821d70b054f25a/lerobot/__init__.py#L94) should serve as the reference for what's working with what.

Thanks for your answer,  I added the modification code in modeling_diffusion.py under #149 but still got the key error problem in configuration_diffusion.py. It seems like only apply Diffusion to pusht 😭

Joeland4, 2024-05-16:
Another problem，when training pusht using diffusion: python lerobot/scripts/train.py     policy=diffusion     env=pusht     env.task=PushT dataset_repo_id=lerobot/pusht
  File "/data1/whx/lerobot/lerobot/scripts/train.py", line 362, in evaluate_and_checkpoint_if_needed
    logger.save_model(
  File "/home/whx/.conda/envs/lerobot/lib/python3.10/site-packages/lerobot/common/logger.py", line 86, in save_model
    artifact = self._wandb.Artifact(
  File "/home/whx/.conda/envs/lerobot/lib/python3.10/site-packages/wandb/sdk/artifacts/artifact.py", line 133, in __init__
    raise ValueError(
ValueError: Artifact name may only contain alphanumeric characters, dashes, underscores, and dots. Invalid name: policy_diffusion-dataset_lerobot/pusht-env_pusht-seed_100000-100000-005000

but no problem for  ACT. @aliberts

aliberts, 2024-05-16:
> ValueError: Artifact name may only contain alphanumeric characters, dashes, underscores, and dots. Invalid name: policy_diffusion-dataset_lerobot/pusht-env_pusht-seed_100000-100000-005000

Did you pull from main recently?
That issue has been solved by #172

alexander-soare, 2024-05-16:
@Joeland4 please let us know if https://github.com/huggingface/lerobot/pull/149 has resolved your main issue here.

Joeland4, 2024-05-17:
> > ValueError: Artifact name may only contain alphanumeric characters, dashes, underscores, and dots. Invalid name: policy_diffusion-dataset_lerobot/pusht-env_pusht-seed_100000-100000-005000
> 
> Did you pull from main recently? That issue has been solved by #172

Thank you!  I should look at all the issues carefully first. 😳

alexander-soare, 2024-05-17:
Glad it's solved!

closed 2024-05-17 by alexander-soare

linCC111, 2024-11-06:
what is the best success rate using Diffusion policy in aloha task?I try 200000 steps in sim_transfer_cube task, but the success rate is about 10%.I want to know whether tring more parameters to train is advisable?