huggingface/lerobot #187: --repo-id for mobile aloha datasets

ramkumarkoppu, 2024-05-16:
most of the aloha datasets in https://huggingface.co/lerobot are not available with the --repo-id argument of lerobot/scripts/visualize_dataset.py
for example how to visualise lerobot/aloha_mobile_wash_pan dataset using this script?

Cadene, 2024-05-16:
I am working on a PR to fix this. I will get back to you as soon as possible

Cadene, 2024-05-20:
Hey @ramkumarkoppu ,the PR is merged. I made sure to fix this issue. Could you check on our side if the issue is solved? Dont hesitate to message me if you need any help.
Best

Cadene, 2024-05-21:
@ramkumarkoppu In this error message I see revision v1.3, but the revision v1.4 is now selected automatically on master.
This could mean that you didnt install the latest version or you didnt checkout master or rebased on master.

Try this:
```bash
git fetch origin master
git checkout origin/master
python lerobot/scripts/visualize_dataset.py --repo-id lerobot/aloha_mobile_wash_pan --episode-index 0
```

closed 2024-05-23 by Cadene

ramkumarkoppu, 2024-05-24:
Hi @Cadene: Sorry for the late reply, it worked 
[image]
after pulling latest, I forgot to run `pip install .` again in my environment to update my python packages versions.