huggingface/lerobot #211: lerobot/common/policies/normalize.py divide by zero

ruijie-he, 2024-05-25:
### System Info

```Shell
Normalize class could have a divide by 0 problem, e.g. when some of the states or actions have a fixed value. 

This can be fixed if https://github.com/huggingface/lerobot/blob/main/lerobot/common/policies/normalize.py#L150 is changed to 

batch[key] = (batch[key] - min) / (max - min + 1e-8)
```

### Information

- [ ] One of the scripts in the examples/ folder of LeRobot
- [X] My own task or dataset (give details below)

### Reproduction

This occurs when I extend push T / Diffusion policy to a real robot, and start by fixing the non-XY states / actions.

### Expected behavior

Should be able to compute loss even when there are states and actions that have a single fixed value in the entire dataset

alexander-soare, 2024-05-27:
@ruijie-he thanks a lot for reporting. We'll add this to the TODO list, but one question: have you considered contributing a PR? ;)

Git-bot9000, 2024-05-27:
@ruijie-he if you are not working on this, can I take a stab at it?

PR #239 (2024-06-04): Fix bug in normalize to avoid divide by zero
## What this does
Fixes #211. Normalize class could have a divide by 0 problem, e.g. when some of the states or actions have a fixed value. 

## How it was tested
Tested after extending push T / Diffusion policy to a real robot and fixing the non-XY states / actions.

- - -
This change is [[image]](https://reviewable.io/reviews/huggingface/lerobot/239)
closed 2024-06-04 by Cadene