huggingface/lerobot #238: how to use  on  wslcan not  visulize

jackylee1, 2024-06-02:
how to use  on  wslcan not  visulize

Cadene, 2024-06-03:
Hello @jackylee1, Really sorry but we didnt understand your message!
Please edit your issue or we might close it in a few days :/
Best regards

jackylee1, 2024-06-05:
ptr = ctx.call(mujoco.MjrContext, model.ptr, font_scale)
  File "/root/anaconda3/envs/lerobot/lib/python3.10/site-packages/dm_control/_render/executor/render_executor.py", line 138, in call
    return func(*args, **kwargs)
mujoco.FatalError: an OpenGL platform library has not been loaded into this process, this most likely means that a valid OpenGL context has not been created before mjr_makeContext was called

Set the environment variable HYDRA_FULL_ERROR=1 for a complete stack trace.

Cadene, 2024-06-05:
@jackylee1 We provide examples of the library that you need to install with `apt-get` (on Linux/Ubuntu) to make mujoco work with OpenGL.

https://github.com/huggingface/lerobot/blob/main/docker/lerobot-gpu-dev/Dockerfile#L8-L40

Let me know if it was useful.

Cadene, 2024-06-05:
@aliberts Should we add this in the README?
I only see `sudo apt-get install cmake build-essential`

jackylee1, 2024-06-05:
raise ImportError('Cannot initialize a headless EGL display.')
ImportError: Cannot initialize a headless EGL display.
 still shows error

jackylee1, 2024-06-05:
please,how to run the docker images ,

aliberts, 2024-06-06:
Hi, yes you need to `export MUJOCO_GL=egl` (no quotes) or simply prepend your python command with `MUJOCO_GL=egl`

We were thinking of a better way of handling in #55 this but I haven't found a way to ensure it's working on any platform for now. Leaving this to the user is probably best for now.

jackylee1, 2024-06-07:
Found the latest cached dataset configuration 'default' at /root/.cache/huggingface/datasets/lerobot___aloha_sim_insertion_human/default/0.0.0/3cdeec058acd3e3583e9c22c24e6c3338b8cd712 (last modified on Sat Jun  1 22:24:10 2024).
WARNING 2024-06-07 14:09:17 che/cache.py:95 Found the latest cached dataset configuration 'default' at /root/.cache/huggingface/datasets/lerobot___aloha_sim_insertion_human/default/0.0.0/3cdeec058acd3e3583e9c22c24e6c3338b8cd712 (last modified on Sat Jun  1 22:24:10 2024).

INFO 2024-06-07 14:31:02 ts/train.py:294 make_env
INFO 2024-06-07 14:31:02 /__init__.py:84 MUJOCO_GL=%s, attempting to import specified OpenGL backend.
INFO 2024-06-07 14:31:02 pesloader.py:70 Failed to load library ( %r ): %s
INFO 2024-06-07 14:31:02 pesloader.py:70 Failed to load library ( %r ): %s
INFO 2024-06-07 14:31:02 tesupport.py:23 No OpenGL_accelerate module loaded: %s
Error executing job with overrides: ['policy=act', 'env=aloha', 'env.task=AlohaInsertion-v0', 'dataset_repo_id=lerobot/aloha_sim_insertion_human']
Traceback (most recent call last):
  File "/opt/lerobot/lerobot/scripts/train.py", line 415, in train_cli
    train(
  File "/opt/lerobot/lerobot/scripts/train.py", line 295, in train
    eval_env = make_env(cfg)
  File "/root/anaconda3/envs/lerobot/lib/python3.10/site-packages/lerobot/common/envs/factory.py", line 48, in make_env
    env = env_cls(
  File "/root/anaconda3/envs/lerobot/lib/python3.10/site-packages/gymnasium/vector/sync_vector_env.py", line 53, in __init__
    self.envs = [env_fn() for env_fn in env_fns]
  File "/root/anaconda3/envs/lerobot/lib/python3.10/site-packages/gymnasium/vector/sync_vector_env.py", line 53, in <listcomp>
    self.envs = [env_fn() for env_fn in env_fns]
  File "/root/anaconda3/envs/lerobot/lib/python3.10/site-packages/lerobot/common/envs/factory.py", line 50, in <lambda>
    lambda: gym.make(gym_handle, disable_env_checker=True, **gym_kwgs)
  File "/root/anaconda3/envs/lerobot/lib/python3.10/site-packages/gymnasium/envs/registration.py", line 756, in make
    env_creator = load_env_creator(env_spec.entry_point)
  File "/root/anaconda3/envs/lerobot/lib/python3.10/site-packages/gymnasium/envs/registration.py", line 545, in load_env_creator
    mod = importlib.import_module(mod_name)
  File "/root/anaconda3/envs/lerobot/lib/python3.10/importlib/__init__.py", line 126, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
  File "<frozen importlib._bootstrap>", line 1050, in _gcd_import
  File "<frozen importlib._bootstrap>", line 1027, in _find_and_load
  File "<frozen importlib._bootstrap>", line 1006, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 688, in _load_unlocked
  File "<frozen importlib._bootstrap_external>", line 883, in exec_module
  File "<frozen importlib._bootstrap>", line 241, in _call_with_frames_removed
  File "/root/anaconda3/envs/lerobot/lib/python3.10/site-packages/gym_aloha/env.py", line 3, in <module>
    from dm_control import mujoco
  File "/root/anaconda3/envs/lerobot/lib/python3.10/site-packages/dm_control/mujoco/__init__.py", line 18, in <module>
    from dm_control.mujoco.engine import action_spec
  File "/root/anaconda3/envs/lerobot/lib/python3.10/site-packages/dm_control/mujoco/engine.py", line 41, in <module>
    from dm_control import _render
  File "/root/anaconda3/envs/lerobot/lib/python3.10/site-packages/dm_control/_render/__init__.py", line 86, in <module>
    Renderer = import_func()
  File "/root/anaconda3/envs/lerobot/lib/python3.10/site-packages/dm_control/_render/__init__.py", line 36, in _import_egl
    from dm_control._render.pyopengl.egl_renderer import EGLContext
  File "/root/anaconda3/envs/lerobot/lib/python3.10/site-packages/dm_control/_render/pyopengl/egl_renderer.py", line 39, in <module>
    from dm_control._render.pyopengl import egl_ext as EGL
  File "/root/anaconda3/envs/lerobot/lib/python3.10/site-packages/dm_control/_render/pyopengl/egl_ext.py", line 29, in <module>
    from OpenGL import EGL
  File "/root/anaconda3/envs/lerobot/lib/python3.10/site-packages/OpenGL/EGL/__init__.py", line 2, in <module>
    from OpenGL.raw.EGL._types import *
  File "/root/anaconda3/envs/lerobot/lib/python3.10/site-packages/OpenGL/raw/EGL/_types.py", line 87, in <module>
    raw_eglQueryString = _p.PLATFORM.EGL.eglQueryString
AttributeError: 'NoneType' object has no attribute 'eglQueryString'

Set the environment variable HYDRA_FULL_ERROR=1 for a complete stack trace.

still shows  error

aliberts, 2024-06-07:
Can you please provide me with more context?

What's your system? (OS, hardware, etc.)
You can use this script for that:
```bash
python lerobot/scripts/display_sys_info.py
```
Are you using a docker image, and if so which one?
Also can you tell what command are you trying to run?

jackylee1, 2024-06-07:
i am on wsl

(lerobot) root@MS-TGCPQOCCPPUG:/opt/lerobot#  python lerobot/scripts/display_sys_info.py

Copy-and-paste the text below in your GitHub issue and FILL OUT the two last points.

- `lerobot` version: 0.1.0
- Platform: Linux-5.15.133.1-microsoft-standard-WSL2-x86_64-with-glibc2.35
- Python version: 3.10.14
- Huggingface_hub version: 0.23.2
- Numpy version: 1.26.4
- PyTorch version (GPU?): 2.3.0+cu121 (True)
- Cuda version: 12010
- Using GPU in script?: <fill in>
- Using distributed or parallel set-up in script?: <fill in>

(lerobot) root@MS-TGCPQOCCPPUG:/opt/lerobot#

aliberts, 2024-06-08:
Can you run try to the gpu docker image?
```bash
docker run -ti --gpus all --shm-size "16gb" huggingface/lerobot-gpu:latest
```
Then inside it:
```bash
DATA_DIR=tests/data pytest -v tests/
```

jackylee1, 2024-06-08:
PS D:\> cd D:\Ubuntu_WSL
PS D:\Ubuntu_WSL> cd lerobotdocker
PS D:\Ubuntu_WSL\lerobotdocker> docker run -ti --gpus all --shm-size "16gb" huggingface/lerobot-gpu:latest
docker: error during connect: in the default daemon configuration on Windows, the docker client must be run with elevated privileges to connect: Head "http://%2F%2F.%2Fpipe%2Fdocker_engine/_ping": open //./pipe/docker_engine: The system cannot find the file specified.
See 'docker run --help'.
PS D:\Ubuntu_WSL\lerobotdocker>
PS D:\Ubuntu_WSL\lerobotdocker> docker run -ti --gpus all --shm-size "16gb" huggingface/lerobot-gpu:latest
Unable to find image 'huggingface/lerobot-gpu:latest' locally
latest: Pulling from huggingface/lerobot-gpu
3c645031de29: Already exists
0d6448aff889: Already exists
0a7674e3e8fe: Already exists
b71b637b97c5: Already exists
56dc85502937: Already exists
4d51d73ad7c8: Pull complete
b9348706a1f3: Pull complete
ed20b7cce72b: Pull complete
0a8fab373494: Pull complete
a2fa069f3555: Pull complete
4f4fb700ef54: Pull complete
20f46d2e78b1: Pull complete
f569dc9eb6fe: Pull complete
Digest: sha256:08f6064c5b7d188de409bf82755be99d51b8062de60c8a0d88cf050bad744698
Status: Downloaded newer image for huggingface/lerobot-gpu:latest
docker: Error response from daemon: failed to create task for container: failed to create shim task: OCI runtime create failed: runc create failed: unable to start container process: error during container init: error running hook #0: error running hook: exit status 1, stdout: , stderr: Auto-detected mode as 'legacy'
nvidia-container-cli: requirement error: unsatisfied condition: cuda>=12.4, please update your driver to a newer version, or use an earlier cuda container: unknown.
PS D:\Ubuntu_WSL\lerobotdocker>

aliberts, 2024-06-08:
Can you build the [Dockerfile](https://github.com/huggingface/lerobot/blob/main/docker/lerobot-gpu/Dockerfile) with your cuda version?
```diff
- FROM nvidia/cuda:12.4.1-base-ubuntu22.04
+ FROM nvidia/cuda:12.1.0-base-ubuntu22.04
```

jackylee1, 2024-06-08:
PS D:\Ubuntu_WSL\perplexicadocker\lerobot-main\docker\lerobot-gpu> docker build -t lerobotcuda12.1.0 .
[+] Building 259.3s (13/13) FINISHED                                                                     docker:default
 => [internal] load build definition from Dockerfile                                                               0.0s
 => => transferring dockerfile: 874B                                                                               0.0s
 => [internal] load metadata for docker.io/nvidia/cuda:12.4.1-base-ubuntu22.04                                     4.6s
 => [internal] load .dockerignore                                                                                  0.0s
 => => transferring context: 2B                                                                                    0.0s
 => [1/9] FROM docker.io/nvidia/cuda:12.4.1-base-ubuntu22.04@sha256:0f6bfcbf267e65123bcc2287e2153dedfc0f24772fb5c  0.1s
 => => resolve docker.io/nvidia/cuda:12.4.1-base-ubuntu22.04@sha256:0f6bfcbf267e65123bcc2287e2153dedfc0f24772fb5c  0.0s
 => => sha256:8767a245ed2c481eb245d8f6c625accc3788e1fb8612403d6b4cd4645a4f09c7 1.37kB / 1.37kB                     0.0s
 => => sha256:ca14dc8401b66a20e1ca678268250834c5c66ac3f458dd570088bb681444ffc0 9.44kB / 9.44kB                     0.0s
 => => sha256:0f6bfcbf267e65123bcc2287e2153dedfc0f24772fb5ce84afe16ac4b2fada95 743B / 743B                         0.0s
 => [internal] load build context                                                                                  0.0s
 => => transferring context: 874B                                                                                  0.0s
 => [2/9] RUN apt-get update && apt-get install -y --no-install-recommends     build-essential cmake     libgli  245.7s
 => [3/9] RUN ln -s /usr/bin/python3.10 /usr/bin/python                                                            0.3s
 => [4/9] RUN python -m venv /opt/venv                                                                             2.2s
 => [5/9] RUN echo "source /opt/venv/bin/activate" >> /root/.bashrc                                                0.5s
 => [6/9] COPY . /lerobot                                                                                          0.0s
 => [7/9] WORKDIR /lerobot                                                                                         0.0s
 => [8/9] RUN pip install --upgrade --no-cache-dir pip                                                             4.4s
 => ERROR [9/9] RUN pip install --no-cache-dir ".[test, aloha, xarm, pusht]"                                       1.5s
------
 > [9/9] RUN pip install --no-cache-dir ".[test, aloha, xarm, pusht]":
0.532 ERROR: Directory '.[test, aloha, xarm, pusht]' is not installable. Neither 'setup.py' nor 'pyproject.toml' found.
------
Dockerfile:26
--------------------
  24 |     WORKDIR /lerobot
  25 |     RUN pip install --upgrade --no-cache-dir pip
  26 | >>> RUN pip install --no-cache-dir ".[test, aloha, xarm, pusht]"
  27 |
  28 |     # Set EGL as the rendering backend for MuJoCo
--------------------
ERROR: failed to solve: process "/bin/sh -c pip install --no-cache-dir \".[test, aloha, xarm, pusht]\"" did not complete successfully: exit code: 1

View build details: docker-desktop://dashboard/build/default/default/hynlno8fd4yum8kwrg4168u7c

Cadene, 2024-06-08:
Sorry, we dont currently provide support for windows. Please install ubuntu on your pc

aliberts, 2024-06-09:
> Directory '.[test, aloha, xarm, pusht]' is not installable. Neither 'setup.py' nor 'pyproject.toml' found.

Yes, you need to build from the root directory of this repo 
(because of `COPY . /lerobot`)

jackylee1, 2024-06-12:
still shows error

closed 2025-10-08 by pkooij