huggingface/lerobot #248: Enable visualisation of local datasets by adding `root` arg

tlpss, 2024-06-05:
I am trying to convert my dataset to a `LeRobotDataset` and have implemented a conversion script. 

I wanted to visualize the dataset locally before uploading or continuing, but the visualization script lacks a `root` arg to enable loading from a local dataset (used in the `LeRobotDataset` init to distinguish between local or remote)

I added this locally and then I was able to visualize the dataset in rerun

[image]

It would be convenient if this would be added to the main branch! 
It is a very small change and I would be willing to make a PR for this.

Cadene, 2024-06-05:
@tlpss Sorry for the lack of documentation. You can already load a LeRobotDataset stored locally with:
```
export DATA_DIR="data"
```
See: https://github.com/huggingface/lerobot/blob/main/lerobot/common/datasets/lerobot_dataset.py#L38-L47

Happy yo add `root` arg in visualize_dataset.py. By any chance, would you have time to put up a PR? Thanks!

tlpss, 2024-06-05:
> By any chance, would you have time to put up a PR? Thanks!

 sure, I'll make a PR for this!

PR #249 (2024-06-10): add root argument to the dataset visualizer to visualize local datasets
## What this does

This PR adds the `root` argument to the dataset visualizer. This parameter is passed to `LeRobotDataset.__init__`,  which enables visualizing local datasets, as an alternative to setting the `DATA_DIR` env variable.

closes #248 

## How it was tested
Explain/show how you tested your changes.

- added a unit test to `tests/test_visualize_dataset` that visualizes the local copy of the `lerobot/pusht` dataset 

## How to checkout & try? (for the reviewer)

Examples:
```
python lerobot/scripts/visualize_dat [... cut]
closed 2024-06-10 by Cadene