huggingface/lerobot #259: when will other models be added ?

ssvision, 2024-06-11:
how do i train and visualise my policy in the environment for other aloha and diffusion tasks? the hugging face repo has only 2 models namely diffusion_pusht & act_aloha_sim_transfer_cube_human?

how do i do the same for other tasks such as aloha insertion task ? and other range of tasks

aliberts, 2024-06-11:
Hi, we are working on adding several models to the library, namely:
- Octo (#209)
- VQ-BEt (#166)
- TD-MPC is already in our codebase and we did reproduce their results initially but since then we've migrated the xarm env and we've been having compatibility issues because of it (the dataset doesn't exactly produces the same outcomes in the new version of the env), which is why we haven't uploaded a pretrained model for it yet.

We'll also be adding more documentation & tutorials on how to switch tasks when training a given policy ;)

alexander-soare, 2024-06-14:
@ssvision if you train a model for a policy/dataset/env combination we haven't already, and it matches or exceeds a SOTA benchmark, we can add it on the hub!
What are you mos interested in? I heard aloha insertion. The reason we haven't added that is that (a) the results are pretty poor on the ACT paper with only 20% success rate, (b) we actually haven't been able to reproduce, falling about 5% short either because of not trying enough seeds or maybe some of these tweaks that still need to go into ACT: https://github.com/huggingface/lerobot/pull/265

alexander-soare, 2024-07-05:
Closing due to inactivity. @ssvision if you have further questions, please don't hesitate to re-open.

closed 2024-07-05 by alexander-soare