huggingface/lerobot #261: Which low cost robot with teleoperation to test the library ?

RochMollero, 2024-06-11:
Firstly, thank you for all the work. At my company we would like to obtain results on real robots from this repository. However, the original setups are either quite expensive (around ~30k for Aloha) or require reconstruction for the UMI interface from Colombia via 3D printing, which would be time-consuming considering we don't have direct experience in the subject.

**Do you have any recommendations for one or more robots with a low-cost teleoperation setup on which we could test and iterate quickly on these algorithms?** I have seen some people doing things with low-cost robots on LinkedIn, and I will reach out to them, but apparently, they do not seem to be selling them.

Thanks,

qgallouedec, 2024-06-11:
Hi! Yes: https://github.com/AlexanderKoch-Koch/low_cost_robot

aliberts, 2024-06-11:
Hi,

Yes the arm from Alexander Koch mentionned by @qgallouedec is a good example of an open source low-cost arm.

We are also experimenting with this one https://github.com/TheRobotStudio/SO-ARM100 from TheRobotStudio, whom we are collaborating with to improve the design.

One of the main differences is that the Koch arm uses Dynamixels motors (which can be hard to source these days) while the SO-ARM100 uses Feetech motors which are cheaper and produce more torque (but are also heavier).

aliberts, 2024-07-23:
Hello there, we now have direct real-world support for the Koch arm with #288 and we'll be adding support for more and more robots in the coming weeks.

Closing this but feel free to reopen if you have any questions ;)

closed 2024-07-23 by aliberts