huggingface/lerobot #289: examples/3_train_policy.py raised a runtime error, saying "tensors must have same numbers of dimensions"

kandeng, 2024-06-22:
### System Info

```Shell
- `lerobot` version: 0.1.0
- Platform: Linux-6.5.0-41-generic-x86_64-with-glibc2.35
- Python version: 3.10.12
- Huggingface_hub version: 0.23.0
- Dataset version: 2.19.1
- Numpy version: 1.26.4
- PyTorch version (GPU?): 2.3.0+cu121 (True)
- Cuda version: 12010
- Using GPU in script?: Yes, 1 * RTX-3070Ti
```

### Information

- [X] One of the scripts in the examples/ folder of LeRobot
- [ ] My own task or dataset (give details below)

### Reproduction

1.  Git clone the entire repo main branch, from github to my laptop with RTX-3070Ti GPU, 

2. Modify https://github.com/huggingface/lerobot/blob/main/examples/3_train_policy.py, as following, 
   Notice that, 
(1)  We tried to train the ACT model,
(2)  We used this training dataset, tests/data/lerobot/aloha_sim_insertion_scripted
(3)  We modified delta_timestamps, to contain  "observation.images.top": [-0.1, 0.0],

~~~
"""
examples/3_train_aloha_policy.py

This scripts demonstrates how to train Diffusion Policy on the PushT environment.

Once you have trained a model with this script, you can try to evaluate it on
examples/2_evaluate_pretrained_policy.py
"""

from pathlib import Path

import torch

from lerobot.common.datasets.lerobot_dataset import LeRobotDataset
from lerobot.common.policies.act.configuration_act import ACTConfig
from lerobot.common.policies.act.modeling_act import ACTPolicy

# Create a directory to store the training checkpoint.
output_directory = Path("outputs/train/example_aloha_act")
output_directory.mkdir(parents=True, exist_ok=True)

# Number of offline training steps (we'll only do offline training for this example.)
# Adjust as you prefer. 5000 steps are needed to get something worth evaluating.
training_steps = 5000
device = torch.device("cuda")
log_freq = 250

# Set up the dataset.
delta_timestamps = {
    # Load the previous image and state at -0.1 seconds before current frame,
    # then load current image and state corresponding to 0.0 second.
    "observation.images.top": [-0.1, 0.0],
    "observation.state": [-0.1, 0.0],
    # Load the previous action (-0.1), the next action to be executed (0.0),
    # and 14 future actions with a 0.1 seconds spacing. All these actions will be
    # used to supervise the policy.
    "action": [-0.1, 0.0, 0.1, 0.2, 0.3, 0.4, 0.5, 0.6, 0.7, 0.8, 0.9, 1.0, 1.1, 1.2, 1.3, 1.4],
}
dataset = LeRobotDataset("lerobot/aloha_sim_insertion_scripted", delta_timestamps=delta_timestamps)

# Set up the the policy.
# Policies are initialized with a configuration class, in this case `DiffusionConfig`.
# For this example, no arguments need to be passed because the defaults are set up for PushT.
# If you're doing something different, you will likely need to change at least some of the defaults.
cfg = ACTConfig()
policy = ACTPolicy(cfg, dataset_stats=dataset.stats)
policy.train()
policy.to(device)

# The code below are NOT changed.   -- Kan, 06.22.2024
optimizer = torch.optim.Adam(policy.parameters(), lr=1e-4)

# Create dataloader for offline training.
dataloader = torch.utils.data.DataLoader(
    dataset,
    num_workers=4,
    batch_size=64,
    shuffle=True,
    pin_memory=device != torch.device("cpu"),
    drop_last=True,
)

# Run training loop.
step = 0
done = False
while not done:
    for batch in dataloader:
        batch = {k: v.to(device, non_blocking=True) for k, v in batch.items()}
        output_dict = policy.forward(batch)
        loss = output_dict["loss"]
        loss.backward()
        optimizer.step()
        optimizer.zero_grad()

        if step % log_freq == 0:
            print(f"step: {step} loss: {loss.item():.3f}")
        step += 1
        if step >= training_steps:
            done = True
            break

# Save a policy checkpoint.
policy.save_pretrained(output_directory)
~~~

3.  Ran the code, and got a runtime error, as following, 

~~~
$ python3 examples/3_train_aloha_policy.py 

 vae_encoder_input[0] type=<class 'torch.Tensor'>, shape=torch.Size([64, 1, 512]) 

 vae_encoder_input[1] type=<class 'torch.Tensor'>, shape=torch.Size([64, 1, 2, 512]) 

 vae_encoder_input[2] type=<class 'torch.Tensor'>, shape=torch.Size([64, 16, 512]) 

Traceback (most recent call last):
  File "/home/robot/lerobot/examples/3_train_aloha_policy.py", line 65, in <module>
    output_dict = policy.forward(batch)
  File "/home/robot/lerobot/lerobot/common/policies/act/modeling_act.py", line 140, in forward
    actions_hat, (mu_hat, log_sigma_x2_hat) = self.model(batch)
  File "/home/robot/.local/lib/python3.10/site-packages/torch/nn/modules/module.py", line 1532, in _wrapped_call_impl
    return self._call_impl(*args, **kwargs)
  File "/home/robot/.local/lib/python3.10/site-packages/torch/nn/modules/module.py", line 1541, in _call_impl
    return forward_call(*args, **kwargs)
  File "/home/robot/lerobot/lerobot/common/policies/act/modeling_act.py", line 315, in forward
    vae_encoder_input = torch.cat(vae_encoder_input, axis=1)
RuntimeError: Tensors must have same number of dimensions: got 3 and 4

~~~

### Expected behavior

Complete the training process successfully without runtime errors.

alexander-soare, 2024-06-25:
@kandeng Right now we don't handle multiple observation steps with ACT. It's probably a fairly easy adjustment to make and we'd welcome a PR if you are interested, and I'd be more than happy to help. Let me know.

kandeng, 2024-06-25:
Many thanks for help, Alex.  @alexander-soare 

No rush to handle multiple observation steps with ACT using [train_policy.py](https://github.com/huggingface/lerobot/blob/main/examples/3_train_policy.py), 

You guys offer another way to train the model, `train.py`, which is easier to follow because of [the well-written user guide](https://github.com/huggingface/lerobot/blob/main/examples/4_train_policy_with_script.md)

For the time being, I think you guys should focus on more important and urgent issues.  

Let me close this issue, and don't let it distract your attention. 

A lot of thanks again, you guys are doing wonderful job!

closed 2024-06-25 by alexander-soare