huggingface/lerobot #293: Clarification Needed on Providing dataset_stats in Diffusion Models

mlkakram, 2024-06-26:
Hi guys, Thank you for all your hard work on this project.

I have a question about using diffusion models and where to provide `dataset_stats`. From the example scripts, it seems that `dataset_stats` should only be provided during model training and are saved with the model using `save_pretrained()`. However, in the file `modeling_diffusion.py`, it's mentioned in the description of the diffusion policy that `dataset_stats` should be provided again using `load_state_dict()`.

I'm a bit confused and would appreciate clarification on the correct practice. Thank you.

alexander-soare, 2024-07-01:
Hi! These statistics need to be present one way or another as long as `config.input_normalization_modes` and/or `config.output_normalization_modes` are provided.

Sometimes its useful to be able to initialise the policy without providing the statistics up front, which is why we have the option not to. That's when you are creating it with the intention of loading in a pretrained model (the statistics are part of the pretrained model params).

Otherwise, you should provide the statistics up front as an argument in the `__init__`.

> dataset_stats should be provided again using load_state_dict().

The key takeaway here is that it doesn't need to be provided "again". One way or the other will do.

Does that help?

alexander-soare, 2024-08-12:
I'm going to close this thread due to inactivity. Please feel free to open it again if needed.

closed 2024-08-12 by alexander-soare