huggingface/lerobot #296: Noob for poetry, missing setup instructions in ReadMe?

PradeepKadubandi, 2024-06-27:
### System Info

```Shell
Perhaps not needed?
```

### Information

- [X] One of the scripts in the examples/ folder of LeRobot
- [ ] My own task or dataset (give details below)

### Reproduction

Try a brand new setup by following the ReadMe step by step.

Open the folder in VS Code. 

Try to debug the 1_load_lerobot_data.py in examples and see import errors and other errors.

More details:
-------------------
The default setup instructions in the ReadMe caused a lot of further package import issues for me when attempting to run the above code. (I created a notebook in the repository to run the script code so that I can iteratively explore the data/code but that's not different from running the script).

The default instructions don't have a step to install the package dependencies needed to run things. I am new to poetry and I see the .toml file specifying dependencies using poetry. But reading poetry documentation felt like reading english poetry - a bit more complicated for simple things like installing dependencies. So I just used pip to manually install package dependencies as I tried to run code in the notebook and saw several import errors (for ex: 'datasets', 'torch', 'torchvision' - all these imports were failing in importing the 'lerobot.common.datasets.lerobot_dataset.LeRobotDataset' class).

It would be useful to add perhaps one instruction from poetry to install the needed dependencies in the ReadMe file so that everyone who's getting started with the repository shouldn't need to go through this and figure this out on their own. 

### Expected behavior

Should be able to get started on running examples smoothly and seamlessly without a lot of hassle and figuring out.

alexander-soare, 2024-07-01:
Hi @PradeepKadubandi, the poetry instructions are just meant for contributors. So you may find them in https://github.com/huggingface/lerobot/blob/main/CONTRIBUTING.md. Can you please take a look and let me know if the instructions there are helpful?

aliberts, 2024-07-01:
Hi, as mentioned above we only use poetry for development. You should be able to run the scripts with everything installed through pip.
Having import errors suggest that something in your install didn't go right.
Can you provide an example of what you're trying to run and the error messages you're getting?

aliberts, 2024-07-23:
Closing this for inactivity, feel free to reopen if need be.

closed 2024-07-23 by aliberts