huggingface/lerobot #297: Add a separate `eval.n_envs` config for setting number of evaluation environments

eugeneteoh, 2024-06-29:
Currently, there are two problems:

- `eval.batch_size` is used for spinning up multiple environments (see [code](https://github.com/huggingface/lerobot/blob/main/lerobot/common/envs/factory.py#L51)). I think this should be renamed to `eval.n_envs` for clarity.
- `eval.batch_size` should be used for computing validation loss.

Related:
https://github.com/huggingface/lerobot/pull/290/files#r1657769267
https://github.com/huggingface/lerobot/issues/250
https://github.com/huggingface/lerobot/pull/283

alexander-soare, 2024-07-01:
Thanks for posting an issue on this @eugeneteoh. Btw what would you call it when we are rolling out in the env, vs when we are cross-validating loss calculation with a holdout set? Right now we have the config `eval` namespace dedicated only to the former.

eugeneteoh, 2024-07-05:
I think keeping it under the `eval` namespace is fine. So it would be `eval.n_envs` and `eval.batch_size`.

closed 2025-10-08 by pkooij