huggingface/lerobot #304: Is there any documentation to create a custom dataset?

HiroIshida, 2024-07-04:
`lerobot/examples` elaborates on how to load and train using the existing dataset on the hugging face repos.
Rather I'd like to know how to turn self-collected data into the dataset. 
So, I'd like to if there is some documentation for that.

RochMollero, 2024-07-09:
Same here. I have my data ready but the datasets class seem rather complex to instantiate, so one (or more depending on the number of camera for examples) examples would be very nice.

TheArtificialOutsider, 2024-07-09:
same here！I really need an example.

zwbx, 2024-07-10:
same question!

Cadene, 2024-07-10:
@TheArtificialOutsider @zwbx @RochMollero Got it! We will address this issue very soon.

Any chance you could provide a very short sample of the datasets in the comment?

In the meantime, a few pointers and ressources:

README:
- https://github.com/huggingface/lerobot?tab=readme-ov-file#the-lerobotdataset-format

See how we use `from_preloaded`:
- https://github.com/huggingface/lerobot/blob/01f8cede0b5f1c16330205b35f4391939e11cb3e/lerobot/scripts/control_robot.py#L336-L344 (see `dataset.stats = stats` to use it directly after)
- https://github.com/huggingface/lerobot/blob/main/lerobot/scripts/push_dataset_to_hub.py#L247-L258

See the content of these files to instantiate the hf_dataset, encode the videos, or store frames, etc.
- from hdf5: https://github.com/huggingface/lerobot/blob/main/lerobot/common/datasets/push_dataset_to_hub/aloha_hdf5_format.py#L209-L211
- from zarr: https://github.com/huggingface/lerobot/blob/main/lerobot/common/datasets/push_dataset_to_hub/pusht_zarr_format.py#L257-L259
- from parquet: https://github.com/huggingface/lerobot/blob/main/lerobot/common/datasets/push_dataset_to_hub/dora_parquet_format.py#L213-L215
- from pickle: https://github.com/huggingface/lerobot/blob/main/lerobot/common/datasets/push_dataset_to_hub/xarm_pkl_format.py#L176-L178

Cadene, 2024-07-10:
cc @michel-aractingi for visibility ;)

zwbx, 2024-07-11:
> CAUTION: External email. Only click on links or open attachments from trusted senders.

Hi, thanks for your attention to this matter. I am using RLbench dataset now. I have raw data now, containing image observations, actions. how can I organize them and transfer them to the hf dataset?

x2ss, 2024-08-27:
It would be even better if there were tutorials on how to train using custom data from the gym simulation environment.

Cadene, 2024-08-27:
Hey there, we are still working on a simplification of the dataset class + upload to hub + tutorial! :)
Sorry if it's taking some time!

Unfortunately, the only option now is to get familiar with: https://github.com/huggingface/lerobot/blob/main/lerobot/scripts/push_dataset_to_hub.py

See some example commands in header. You can eventually adapt one of them to your dataset format. 
If you have issue understanding the code, reach out to us on discord `#help` channel

aliberts, 2024-09-02:
Closing this as we now have a [tutorial](https://github.com/huggingface/lerobot/blob/main/examples/7_get_started_with_real_robot.md) to easily record and push your own dataset.
Feel free to reopen if need be ;)

closed 2024-09-02 by aliberts

x2ss, 2024-09-03:
As said in the tutorial “If you don't want to push to hub, use --push-to-hub 0.”, where --push-to-hub 0 should be use?  Replace --repo-id ${HF_USER}/koch_test \ with --push-to-hub 0?

Thanks and best regards!

aliberts, 2024-09-03:
`--push-to-hub 0` is an option of the [`lerobot/scripts/control_robot.py`](https://github.com/huggingface/lerobot/blob/main/lerobot/scripts/control_robot.py) script. This is simply to deactivate uploading your dataset to the hub when using the `record` function.

As for the tutorial, it teaches you — amongst other things — how to record a LeRobotDataset with the [Koch v1.1 arm](https://github.com/jess-moss/koch-v1-1) (although it can be adapted to other robots, we are working on it).

> It seems to me that a useful example would be as [ACT](https://github.com/tonyzhaozh/act/blob/main/record_sim_episodes.py#L161), where the dataset is saved in a clear way.

Could you elaborate? What other scenario do you have in mind?

FANG-Zhiwei, 2025-06-18:
> Closing this as we now have a [tutorial](https://github.com/huggingface/lerobot/blob/main/examples/7_get_started_with_real_robot.md) to easily record and push your own datasets. Feel free to reopen if need be ;)

Hi! It seems that this file is missing. I'm still looking for a tutorial to guide how to record a custom dataset with robots out of officially supported ones (franka panda to be specifically). Could you please suggest some material?

tc-huang, 2025-06-18:
> Hi! It seems that this file is missing. I'm still looking for a tutorial to guide how to record a custom dataset with robots out of officially supported ones (franka panda to be specifically). Could [...]

Hello @FANG-Zhiwei,

I'm not sure if the documents [Record a dataset](https://huggingface.co/docs/lerobot/il_robots#record-a-dataset) and [Bring Your Own Hardware](https://huggingface.co/docs/lerobot/integrate_hardware) are what you're looking for, but you can check them out.

Hope this helps!