huggingface/lerobot #305: how to infer the policy trained by lerobot in real env?

cong1024, 2024-07-05:
### System Info

```Shell
how to infer the policy trained by lerobot in real env?
```

### Information

- [ ] One of the scripts in the examples/ folder of LeRobot
- [ ] My own task or dataset (give details below)

### Reproduction

in the code, i have not found any solution to transfer policy rollout to real env, please help me figure it out

### Expected behavior

how to infer the policy trained by lerobot in real env?

Cadene, 2024-07-06:
@cong1024 We are working on a PR + tutorials: https://github.com/huggingface/lerobot/pull/288

cong1024, 2024-07-08:
thanks a lot!

ysyyork, 2024-07-09:
@Cadene this is awesome! look forward to see the work merged

cong1024, 2024-07-17:
Nice, I see the pr has been merged,  by now only support the robot koch, will there be more support for other type of robot for example aloha or unitree or  so-100 robot?

aliberts, 2024-07-23:
> @Cadene Nice, I see the pr has been merged, by now only support the robot koch, will there be more support for other type of robot for example aloha or unitree or so-100 robot in future? thx

Hi there, yes we are working on adding support for more robots in the near future, including SO-100 and Aloha ;)

I'm closing this in the meantime, feel free to reopen if needed.

closed 2024-07-23 by aliberts