huggingface/lerobot #311: a question about evaluation in sim env

cong1024, 2024-07-09:
### System Info

```Shell
ubuntu 20
```

### Information

- [ ] One of the scripts in the examples/ folder of LeRobot
- [ ] My own task or dataset (give details below)

### Reproduction

1.transform the  real world collected data (hdf5 format) into mp4 format using transform scripts.
2.training with lerobot framework (act , aloha) with 20W steps
3. eval in sim env（gym aloha）, however, the robotic arm hardly moves and shows no tendency to search for objects.
have you guys met the same problem and have some idea for that? 

### Expected behavior

very appreciate for this great work

alexander-soare, 2024-07-09:
Hi! Thanks for reporting.

Can you please provide your training script and the lerobot repo commit hash? (and anything else I might need to reproduce it). What do you mean by 20W steps? What is the eval success metric that you get?

cong1024, 2024-07-10:
@alexander-soare  sure! Thanks for your support,
1. for the training script, I basicly use the original training script,  just modify the default yaml to act yaml, where 20w means 20000 hh.

2, for the dataset, I use the script provided by you guys to transform my real world data to mp4 format

3. For eval, I change the original xml in gym aloha to my  env.

4. For the eval success metric I do not set anything, because i just want to eval if  the robot have the  tendency to search for objects, if it is successful, the next step I will set the success metric

alexander-soare, 2024-08-12:
I'll close this thread as, I believe we picked it up on email. Please feel free to open it again it needed.

closed 2024-08-12 by alexander-soare