huggingface/lerobot #313: Use of dataset stats override

fislamrobo, 2024-07-09:
### System Info

```Shell
Mac OS
```

### Information

- [X] One of the scripts in the examples/ folder of LeRobot
- [ ] My own task or dataset (give details below)

### Reproduction

Wondering why do we use override stats for the policies that do not use pre-trained resnet models eg Diffusion:
configs/policy/diffusion.yaml
Shouldn't these be computed from the dataset?
What do the original authors use?

### Expected behavior

Appreciate your assistance.

alexander-soare, 2024-07-09:
In diffusion policy, these were used to exactly match the stats from the original implementation so that we could port their weights and reproduce eval results. We did this to avoid any differences in stats because of numerical errors associated with the way we compute the stats.

Apart from diffusion policy, we don't use the overrides anywhere else where we do not use pre-trained image models.

Hope that answers your question :)

fislamrobo, 2024-07-09:
Thanks for the prompt response @alexander-soare . In the author's repo, I am looking at:
diffusion_policy/model/vision/multi_image_obs_encoder.py
Where the values they used are: mean=[0.485, 0.456, 0.406], std=[0.229, 0.224, 0.225]
whereas the ones you have are: mean: [[[0.5]], [[0.5]], [[0.5]]]  # (c,1,1), std: [[[0.5]], [[0.5]], [[0.5]]]  # (c,1,1)

alexander-soare, 2024-07-09:
Yeah, I can't remember where exactly but elsewhere it ends up being overridden. You can check this by breakpointing in their code where the normalization happens. In any case, we wouldn't have been able to port weights and eval correctly if we got this wrong.

fislamrobo, 2024-07-09:
@alexander-soare with an std of 0.5 (and with mean of 0.5), the values can go outside of 0 to 1 range. I see one place in the code where the mean is 0.5 but the std is 1/12.
Its in get_image_range_normalizer func in diffusion_policy/common/normalize_util.py

alexander-soare, 2024-07-10:
I see, and what's the issue with the values being outside that range?
Note that you won't be able to find the normalization values from our config in the code. They are either dynamically computed, or loaded from their dataset files.

fislamrobo, 2024-07-12:
Unnormalized values are between 0-1 so I thought it might be undesirable to to have normalized values go outside of this range. I am not sure if it's a real issue though.
To your second point, I was under the impression that these values in yaml override the calculated values.

alexander-soare, 2024-07-15:
> To your second point, I was under the impression that these values in yaml override the calculated values.

Okay maybe I'm getting confused between LeRobot and the official Diffusion Policy repository. Without trying to untangle the details, I can confidently say that we have dataset normalization stats that match theirs for the PushT (image) task, and this can be verified by breakpointing into their code at the right place when eval'ing PushT.

closed 2024-08-12 by alexander-soare