huggingface/lerobot #324: High Data Loading Times at the Beginning of Each Epoch with LeRobot Datasets

rakhimovv, 2024-07-17:
### System Info

```Shell
- `lerobot` version: 0.1.0
- Platform: Linux-6.5.0-41-generic-x86_64-with-glibc2.35
- Python version: 3.10.14
- Huggingface_hub version: 0.23.4
- Dataset version: 2.20.0
- Numpy version: 1.26.4
- PyTorch version (GPU?): 2.3.1+cu121 (True)
- Cuda version: 12010
```

### Information

- [X] One of the scripts in the examples/ folder of LeRobot
- [ ] My own task or dataset (give details below)

### Reproduction

Hello,

I have 10 LeRobot datasets, each containing 50 episodes. I'm looking for an effective way to work with them. Should I use a MultiLeRobot dataset? Or it is better to make a single LeRobot dataset/ How should I handle chunks?

The problem is that data loading time is extremely high at the beginning of each epoch.

For example, when I run the following command:

```bash
python lerobot/scripts/train.py \
    policy=act \
    env=aloha \
    env.task=AlohaInsertion-v0 \
    dataset_repo_id=lerobot/aloha_sim_insertion_human
```

and explicitly set `num_samples` to return 50, the output is:

```plaintext
INFO 2024-07-17 14:20:13 ts/train.py:306 make_env
INFO 2024-07-17 14:20:13 /__init__.py:84 MUJOCO_GL=%s, attempting to import specified OpenGL backend.
INFO 2024-07-17 14:20:13 /__init__.py:31 MuJoCo library version is: %s
INFO 2024-07-17 14:20:40 ts/train.py:309 make_policy
INFO 2024-07-17 14:20:43 on/logger.py:39 Output dir: outputs/train/2024-07-17/14-20-07_aloha_act_default
INFO 2024-07-17 14:20:43 ts/train.py:330 cfg.env.task='AlohaInsertion-v0'
INFO 2024-07-17 14:20:43 ts/train.py:331 cfg.training.offline_steps=100000 (100K)
INFO 2024-07-17 14:20:43 ts/train.py:332 cfg.training.online_steps=0
INFO 2024-07-17 14:20:43 ts/train.py:333 offline_dataset.num_samples=50 (50)
INFO 2024-07-17 14:20:43 ts/train.py:334 offline_dataset.num_episodes=34
INFO 2024-07-17 14:20:43 ts/train.py:335 num_learnable_params=51613582 (52M)
INFO 2024-07-17 14:20:43 ts/train.py:336 num_total_params=51613672 (52M)
INFO 2024-07-17 14:20:44 ts/train.py:401 Start offline training on a fixed dataset
INFO 2024-07-17 14:20:46 ts/train.py:407 Step: 0, dataloading_s: 1.883
INFO 2024-07-17 14:20:54 ts/train.py:184 step:0 smpl:8 ep:5 epch:0.16 loss:68.760 grdn:1310.634 lr:1.0e-05 updt_s:8.240 data_s:1.883
INFO 2024-07-17 14:20:54 ts/train.py:407 Step: 1, dataloading_s: 0.000
INFO 2024-07-17 14:20:54 ts/train.py:407 Step: 2, dataloading_s: 0.000
INFO 2024-07-17 14:20:54 ts/train.py:407 Step: 3, dataloading_s: 0.000
INFO 2024-07-17 14:20:54 ts/train.py:407 Step: 4, dataloading_s: 0.000
INFO 2024-07-17 14:20:55 ts/train.py:407 Step: 5, dataloading_s: 0.000
INFO 2024-07-17 14:20:55 ts/train.py:407 Step: 6, dataloading_s: 0.000
INFO 2024-07-17 14:20:58 ts/train.py:407 Step: 7, dataloading_s: 1.996
INFO 2024-07-17 14:20:59 ts/train.py:407 Step: 8, dataloading_s: 0.000
INFO 2024-07-17 14:20:59 ts/train.py:407 Step: 9, dataloading_s: 0.000
INFO 2024-07-17 14:20:59 ts/train.py:407 Step: 10, dataloading_s: 0.000
INFO 2024-07-17 14:20:59 ts/train.py:407 Step: 11, dataloading_s: 0.405
INFO 2024-07-17 14:21:00 ts/train.py:407 Step: 12, dataloading_s: 0.000
INFO 2024-07-17 14:21:00 ts/train.py:407 Step: 13, dataloading_s: 0.000
INFO 2024-07-17 14:21:02 ts/train.py:407 Step: 14, dataloading_s: 1.966
INFO 2024-07-17 14:21:02 ts/train.py:407 Step: 15, dataloading_s: 0.000
INFO 2024-07-17 14:21:02 ts/train.py:407 Step: 16, dataloading_s: 0.000
INFO 2024-07-17 14:21:02 ts/train.py:407 Step: 17, dataloading_s: 0.000
INFO 2024-07-17 14:21:03 ts/train.py:407 Step: 18, dataloading_s: 0.432
INFO 2024-07-17 14:21:03 ts/train.py:407 Step: 19, dataloading_s: 0.000
INFO 2024-07-17 14:21:03 ts/train.py:407 Step: 20, dataloading_s: 0.000
INFO 2024-07-17 14:21:05 ts/train.py:407 Step: 21, dataloading_s: 1.913
INFO 2024-07-17 14:21:05 ts/train.py:407 Step: 22, dataloading_s: 0.000
INFO 2024-07-17 14:21:06 ts/train.py:407 Step: 23, dataloading_s: 0.000
INFO 2024-07-17 14:21:06 ts/train.py:407 Step: 24, dataloading_s: 0.000
INFO 2024-07-17 14:21:06 ts/train.py:407 Step: 25, dataloading_s: 0.380
INFO 2024-07-17 14:21:07 ts/train.py:407 Step: 26, dataloading_s: 0.000
INFO 2024-07-17 14:21:07 ts/train.py:407 Step: 27, dataloading_s: 0.000
INFO 2024-07-17 14:21:09 ts/train.py:407 Step: 28, dataloading_s: 1.819
INFO 2024-07-17 14:21:09 ts/train.py:407 Step: 29, dataloading_s: 0.001
INFO 2024-07-17 14:21:09 ts/train.py:407 Step: 30, dataloading_s: 0.000
INFO 2024-07-17 14:21:09 ts/train.py:407 Step: 31, dataloading_s: 0.000
INFO 2024-07-17 14:21:10 ts/train.py:407 Step: 32, dataloading_s: 0.348
INFO 2024-07-17 14:21:10 ts/train.py:407 Step: 33, dataloading_s: 0.000
INFO 2024-07-17 14:21:10 ts/train.py:407 Step: 34, dataloading_s: 0.000
INFO 2024-07-17 14:21:12 ts/train.py:407 Step: 35, dataloading_s: 1.846
INFO 2024-07-17 14:21:12 ts/train.py:407 Step: 36, dataloading_s: 0.000
INFO 2024-07-17 14:21:12 ts/train.py:407 Step: 37, dataloading_s: 0.000
INFO 2024-07-17 14:21:13 ts/train.py:407 Step: 38, dataloading_s: 0.000
INFO 2024-07-17 14:21:13 ts/train.py:407 Step: 39, dataloading_s: 0.388
INFO 2024-07-17 14:21:13 ts/train.py:407 Step: 40, dataloading_s: 0.000
INFO 2024-07-17 14:21:13 ts/train.py:407 Step: 41, dataloading_s: 0.000
```

As you can see, the first step at the beginning of each epoch (steps 0, 7, 14, etc., depending on batch size) takes a long time to load. For custom datasets with larger image sizes, the situation worsens, with the first iteration taking up to 10 seconds.

Any suggestions on how to address this issue would be greatly appreciated.

### Expected behavior

Consistent data loading times across all steps, including the first, to prevent GPU utilization from dropping to zero at the beginning of each epoch.

alexander-soare, 2024-07-17:
I think you can use `persistent_workers=True` in the dataloader. We don't plumb that option through, but can you please try it by  putting it in the training script?
Also the `dataloading_s` is inconsistent between steps because the dataloader works by prefetching batches, so when it's 0 it means its just giving you a prefetched batch and when it's > 0 it means it is working on prefetching batches.

cc @Cadene as this slowness is also likely due to accessing HF datasets.

rakhimovv, 2024-07-18:
@alexander-soare Thanks, I tried `persistent_workers=True`, no effect.

Found a similar issue and solution here https://github.com/huggingface/pytorch-image-models/pull/140

Made a pull request, now there is no lag :)

jadechoghari, 2025-10-08:
We now have a new version of lerobotdataset v3.0

closed 2025-10-08 by jadechoghari