huggingface/lerobot #342: Try to run visualize dataset script and fail on the distant linux machine which I ssh in through vscode from windows

luh-j, 2024-07-26:
I'm encountering the problem on the distant linux machine which I ssh in through VScode, and when I try to run the script 
```bash
python lerobot/scripts/visualize_dataset.py \
    --repo-id lerobot/pusht \
    --episode-index 0
```
The following issue occurred:
```bash
Fetching 212 files: 100%|██████████████████████████████████████████████████████████████████| 212/212 [00:00<00:00, 5920.88it/s]
[2024-07-26T09:47:23Z INFO  re_sdk_comms::server] Hosting a SDK server over TCP at 0.0.0.0:9876. Connect with the Rerun logging SDK.
Error: winit EventLoopError: os error at /usr/local/cargo/registry/src/index.crates.io-6f17d22bba15001f/winit-0.29.15/src/platform_impl/linux/mod.rs:786: neither WAYLAND_DISPLAY nor WAYLAND_SOCKET nor DISPLAY is set.
[2024-07-26T09:47:24Z WARN  re_sdk_comms::buffered_client] Failed to send message after 3 attempts: Failed to connect to Rerun server at 127.0.0.1:9876: Connection refused (os error 111)
[2024-07-26T09:47:27Z WARN  re_sdk_comms::buffered_client] Dropping messages because tcp client has timed out.
[2024-07-26T09:47:27Z WARN  re_sdk_comms::buffered_client] Dropping messages because tcp client has timed out.
[2024-07-26T09:47:27Z WARN  re_sdk_comms::tcp_client] Tried to flush while TCP stream was still Pending. Data was possibly dropped.
100%|████████████████████████████████████████████████████████████████████████████████████████████| 6/6 [00:00<00:00,  9.43it/s]
```
I would love to run the visualize dataset script successfully on the ssh linux machine, and I'm wondering if there's anything I can do next. 

System info:
- `lerobot` version: 0.1.0
- Platform: Linux-5.15.0-91-generic-x86_64-with-glibc2.35
- Python version: 3.10.14
- Huggingface_hub version: 0.24.2
- Dataset version: 2.20.0
- Numpy version: 1.26.4
- PyTorch version (GPU?): 2.4.0+cu121 (True)
- Cuda version: 12010
- Using GPU in script?: <fill in>

Cadene, 2024-07-27:
We have an alternative that uses flask: https://github.com/huggingface/lerobot/pull/188

We are still iterating a bit but it works pretty well already: 
https://huggingface.co/spaces/lerobot/visualize_all_datasets

Cadene, 2024-08-12:
@luh-j You can now use:
```bash
python lerobot/scripts/visualize_dataset_html.py \
    --repo-id lerobot/pusht \
    --episode-indices 0
```

We will add a tutorial soon.

Closing :)

closed 2024-08-12 by Cadene