huggingface/lerobot #368: Question: adam optimizer with weight decay for diffusion policy implementation

leonmkim, 2024-08-20:
Thank you for this incredibly useful repo! I had a small question regarding the optimizer used for training diffusion policies as it seems like Adam is used in this implementation, but glancing at the DP author's codebase for both [DP](https://github.com/real-stanford/diffusion_policy/blob/548a52bbb105518058e27bf34dcf90bf6f73681a/diffusion_policy/config/train_diffusion_unet_image_workspace.yaml#L88) and [UMI](https://github.com/real-stanford/universal_manipulation_interface/blob/298776ce251f33b6b3185a98d6e7d1f9ad49168b/diffusion_policy/workspace/train_diffusion_unet_image_workspace.py#L77), it seems they use AdamW. As far as I know, pytorch's Adam and AdamW handle weight decay differently and so I was wondering if this was an intentional deviation from the original implementation. 

Apologies if I misread anything and thanks again.

alexander-soare, 2024-08-21:
Thanks so much for reporting this. Oops... this was my mistake. It was not intentional. Now that you ask, do you believe there's a strong reason to switch back to AdamW? (other than that it's how the original does it?)

leonmkim, 2024-08-21:
Other than for reproducibility using the default hparams, I personally don't know. Someone else, perhaps the authors, may have the evidence for how sensitive diffusion policy is to the choice of optimizer/weight decay parameter.  Happy to have the issue closed as I was more curious than anything. 

Thanks for the response!

closed 2024-08-22 by alexander-soare