huggingface/lerobot #405: ERROR: Could not build wheels for mujoco, which is required to install pyproject.toml-based projects

emozhn, 2024-09-03:
### System Info

```Shell
ERROR: Could not build wheels for mujoco, which is required to install pyproject.toml-based projects. This is the ERROR when I run " pip install -e ".[aloha, pusht]"
```

### Information

- [X] One of the scripts in the examples/ folder of LeRobot
- [ ] My own task or dataset (give details below)

### Reproduction

Building wheels for collected packages: lerobot, mujoco
  Building editable for lerobot (pyproject.toml) ... done
  Created wheel for lerobot: filename=lerobot-0.1.0-py3-none-any.whl size=15433 sha256=fd9b4e6cf834d5059fb230a7da58e72e41286d4108c972d18f0964ceb36e3be7
  Stored in directory: C:\Users\Nan Zhou\AppData\Local\Temp\pip-ephem-wheel-cache-gos3nxx6\wheels\20\b8\41\2023316fdd090518b06150d21f97ccdf9353f7eb05daa64f59
  Building wheel for mujoco (setup.py) ... error
  error: subprocess-exited-with-error

  × python setup.py bdist_wheel did not run successfully.
  │ exit code: 1
  ╰─> [68 lines of output]
      running bdist_wheel
      running build
      running build_py
      creating build
      creating build\lib.win-amd64-cpython-312
      creating build\lib.win-amd64-cpython-312\mujoco
      copying mujoco\bindings_test.py -> build\lib.win-amd64-cpython-312\mujoco
      copying mujoco\gl_context.py -> build\lib.win-amd64-cpython-312\mujoco
      copying mujoco\renderer.py -> build\lib.win-amd64-cpython-312\mujoco
      copying mujoco\renderer_test.py -> build\lib.win-amd64-cpython-312\mujoco
      copying mujoco\render_test.py -> build\lib.win-amd64-cpython-312\mujoco
      copying mujoco\rollout.py -> build\lib.win-amd64-cpython-312\mujoco
      copying mujoco\rollout_test.py -> build\lib.win-amd64-cpython-312\mujoco
      copying mujoco\viewer.py -> build\lib.win-amd64-cpython-312\mujoco
      copying mujoco\viewer_test.py -> build\lib.win-amd64-cpython-312\mujoco
      copying mujoco\__init__.py -> build\lib.win-amd64-cpython-312\mujoco
      creating build\lib.win-amd64-cpython-312\mujoco\cgl
      copying mujoco\cgl\cgl.py -> build\lib.win-amd64-cpython-312\mujoco\cgl
      copying mujoco\cgl\__init__.py -> build\lib.win-amd64-cpython-312\mujoco\cgl
      creating build\lib.win-amd64-cpython-312\mujoco\egl
      copying mujoco\egl\egl_ext.py -> build\lib.win-amd64-cpython-312\mujoco\egl
      copying mujoco\egl\__init__.py -> build\lib.win-amd64-cpython-312\mujoco\egl
      creating build\lib.win-amd64-cpython-312\mujoco\glfw
      copying mujoco\glfw\__init__.py -> build\lib.win-amd64-cpython-312\mujoco\glfw
      creating build\lib.win-amd64-cpython-312\mujoco\osmesa
      copying mujoco\osmesa\__init__.py -> build\lib.win-amd64-cpython-312\mujoco\osmesa
      running build_ext
      Traceback (most recent call last):
        File "<string>", line 2, in <module>
        File "<pip-setuptools-caller>", line 34, in <module>
        File "C:\Users\Nan Zhou\AppData\Local\Temp\pip-install-ta035g8n\mujoco_11cff27e5e4f4f54bc8fbb10cf504c1b\setup.py", line 418, in <module>
          setup(**SETUP_KWARGS)
        File "C:\Users\Nan Zhou\.conda\envs\pr\Lib\site-packages\setuptools\__init__.py", line 103, in setup
          return distutils.core.setup(**attrs)
                 ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
        File "C:\Users\Nan Zhou\.conda\envs\pr\Lib\site-packages\setuptools\_distutils\core.py", line 184, in setup
          return run_commands(dist)
                 ^^^^^^^^^^^^^^^^^^
        File "C:\Users\Nan Zhou\.conda\envs\pr\Lib\site-packages\setuptools\_distutils\core.py", line 200, in run_commands
          dist.run_commands()
        File "C:\Users\Nan Zhou\.conda\envs\pr\Lib\site-packages\setuptools\_distutils\dist.py", line 969, in run_commands
          self.run_command(cmd)
        File "C:\Users\Nan Zhou\.conda\envs\pr\Lib\site-packages\setuptools\dist.py", line 976, in run_command
          super().run_command(command)
        File "C:\Users\Nan Zhou\.conda\envs\pr\Lib\site-packages\setuptools\_distutils\dist.py", line 988, in run_command
          cmd_obj.run()
        File "C:\Users\Nan Zhou\.conda\envs\pr\Lib\site-packages\setuptools\command\bdist_wheel.py", line 373, in run
          self.run_command("build")
        File "C:\Users\Nan Zhou\.conda\envs\pr\Lib\site-packages\setuptools\_distutils\cmd.py", line 316, in run_command
          self.distribution.run_command(command)
        File "C:\Users\Nan Zhou\.conda\envs\pr\Lib\site-packages\setuptools\dist.py", line 976, in run_command
          super().run_command(command)
        File "C:\Users\Nan Zhou\.conda\envs\pr\Lib\site-packages\setuptools\_distutils\dist.py", line 988, in run_command
          cmd_obj.run()
        File "C:\Users\Nan Zhou\.conda\envs\pr\Lib\site-packages\setuptools\_distutils\command\build.py", line 132, in run
          self.run_command(cmd_name)
        File "C:\Users\Nan Zhou\.conda\envs\pr\Lib\site-packages\setuptools\_distutils\cmd.py", line 316, in run_command
          self.distribution.run_command(command)
        File "C:\Users\Nan Zhou\.conda\envs\pr\Lib\site-packages\setuptools\dist.py", line 976, in run_command
          super().run_command(command)
        File "C:\Users\Nan Zhou\.conda\envs\pr\Lib\site-packages\setuptools\_distutils\dist.py", line 988, in run_command
          cmd_obj.run()
        File "C:\Users\Nan Zhou\AppData\Local\Temp\pip-install-ta035g8n\mujoco_11cff27e5e4f4f54bc8fbb10cf504c1b\setup.py", line 156, in run
          self._mujoco_framework_path) = self._find_mujoco()
                                         ^^^^^^^^^^^^^^^^^^^
        File "C:\Users\Nan Zhou\AppData\Local\Temp\pip-install-ta035g8n\mujoco_11cff27e5e4f4f54bc8fbb10cf504c1b\setup.py", line 170, in _find_mujoco
          raise RuntimeError(
      RuntimeError: MUJOCO_PATH environment variable is not set
      [end of output]

  note: This error originates from a subprocess, and is likely not a problem with pip.
  ERROR: Failed building wheel for mujoco
  Running setup.py clean for mujoco
Successfully built lerobot
Failed to build mujoco
ERROR: Could not build wheels for mujoco, which is required to install pyproject.toml-based projects

### Expected behavior

Response

aliberts, 2024-09-03:
Hi,
Can you paste here your system info?
```bash
python lerobot/scripts/display_sys_info.py
```
Also, do you get the same if you try `pip install mujoco` in a new conda environment with python 3.10?

bereldhuin, 2025-04-01:
Hi,

Same problem here.
Here is the result of display_sys_info.py : 

```
- `lerobot` version: N/A
- Platform: Linux-6.11.0-21-generic-x86_64-with-glibc2.39
- Python version: 3.12.9
- Huggingface_hub version: N/A
- Dataset version: N/A
- Numpy version: 2.2.4
- PyTorch version (GPU?): N/A (N/A)
- Cuda version: N/A
- Using GPU in script?: <fill in>
```

I also did a pipi install mujoco without success.

Do I need to run it with Pyhton 3.10 instead of 3.12 ?

bereldhuin, 2025-04-01:
Indeed, works better with Python 3.10...

Mystery solved.