huggingface/lerobot #408: Isaac Sim integration with lerobor

SakethTecholution, 2024-09-03:
It would be great to have realistic simulation training using isaacsim and Isaac lab with LeRobot.

Cadene, 2024-09-03:
I agree! We might port it soon with an simple example.

SakethTecholution, 2024-09-03:
@Cadene 
That would be great. ! Even I would like to contribute if anything I can do into this..

Cadene, 2024-09-03:
With pleasure. We would need to port a simulator and a simple task with a gym interface similarly to what we did with PushT (pygame) and Aloha (mujoco)

SakethTecholution, 2024-09-03:
@Cadene 
And I think there should be some tutorials are required for training a real industrial cobot like franka or UR along with LeRobot which could be helpful to make it more standardized and useful

tommywu052, 2025-01-12:
interested to follow up this thread for Isaac training , any update?

apockill, 2025-01-12:
This is tangentially related, but I built a tool for generating synthetic data in Isaac SIM.

It outputs lerobot datasets:
https://github.com/apockill/pi_at_home/

tommywu052, 2025-01-12:
> This is tangentially related, but I built a tool for generating synthetic data in Isaac SIM. I'm happy to help anyone setting it up.
> 
> It outputs lerobot datasets: https://github.com/apockill/pi_at_home/

it looks like an interesting project!  Does that use Isaac sim to record and save to Lerobot dataset format.  How you sync with real robot like so-arm100 , using ROS2 bridge?

apockill, 2025-01-12:
Yep! It provides isaac extensions and an environment to record, replay, and render trajectories with domain randomization / augmentation applied. Those rendered trajectories are then convertable to lerobot.

For new robots, there's a base class you implement to read/write joints for your specific robot, then you specify that in configuration. Next you'd import an so-arm100 (or whatever robot) URDF into isaac sim, and the extension should be compatible with your robot!

jccalvojackson, 2025-01-17:
What other simulations of the so-100 are out there? I saw this one https://github.com/samzapo/gym-drake-lca
mentioned here https://github.com/huggingface/lerobot/pull/437

twerdster, 2025-02-06:
https://wiki.seeedstudio.com/lerobot_so100m_isaacsim/

ameiworking, 2025-04-25:
Hi, I’d appreciate some advice regarding using so-arm100 on IsaacSim.

I'm trying to use the action data from a [lerobot dataset ](https://huggingface.co/spaces/lerobot/visualize_dataset?dataset=youliangtan%2Fso100_strawberry_grape&episode=39)to control the SO100 robot in Isaac Sim. However, it doesn't behave correctly, and I suspect the issue might be due to a format mismatch.

Here's what I’ve done so far:
1. I have followed the instruction to import  so-arm100 and made the ros2 node to send the action.
> https://wiki.seeedstudio.com/lerobot_so100m_isaacsim/ 

2. I assumed the action data in the dataset is in degrees, so I converted it to radians.
Then, I assigned the converted values to JointState.position.
But when I run it in Isaac Sim, the resulting motion is completely off.

Does anyone have any suggestions on what might be wrong or what I should check?

Thanks in advance!

pjw1, 2025-05-20:
Maybe you can use urdf file in here:

https://github.com/TheRobotStudio/SO-ARM100/issues/54

mirkokiefer, 2025-06-15:
Curiuos if anyone made progress on this.

16Huzeyu, 2025-07-15:
Hi！Everyone！We just opensource this！
🚀 LeIsaac — Open-source workflow combining LeRobot  x IsaacSim!

Lightwheel has developed LeIsaac, a complete and reproducible workflow that demonstrates how to fine-tune GR00T N1.5 using data collected entirely from simulation — and deploy it seamlessly on real robots!

💡 What’s included?
 🏠 SimReady assets — Fully prepared kitchen environment, LeRobot SO101, oranges, and more.
 🎮 LeRobot teleoperation data collection in simulation — Integrated end-to-end pipeline for data collection directly in simulation.
 🧠 GR00T N1.5 fine-tuning and deployment — Train GR00T N1.5 with simulation data, then deploy and validate on real hardware. 

⭐ In the demo video below, the entire process is completed in just 5 simple steps.

👉 GITHUB:  https://github.com/LightwheelAI/leisaac

closed 2025-10-08 by pkooij