huggingface/lerobot #413: Compatible off-the-shelf robots?

danielfriis, 2024-09-05:
Huge thanks for making all of this available!

Can you recommend any (low-cost) off-the-shelf robots to work with?

Cadene, 2024-09-05:
We are working on Moss v1 which will be off-the-shelf and more affordable than Koch v1.1

Cadene, 2024-09-06:
Probably mid-October

yarcles, 2024-09-15:
Thank you for all your amazing work.

I'll be ready for a bulk order :)

danielfriis, 2024-09-22:
Would something like this work: https://www.trossenrobotics.com/pincherx100 ?

MichaelrMentele, 2024-10-04:
You can use the pincher pretty trivially the problem is you don't have a controller that works off the shelf.

MichaelrMentele, 2024-10-15:
@danielfriis @yarcles I'm providing a batch of [10 Koch 2.0 arms near cost for early access.](https://www.bananarobotics.ai/product/r0) with code LEROBOT at checkout. 

They come pre-assembled and calibrated -- ready for plug and play use with LeRobot. No questions asked return. Willing to customize material for the first 10 as well!

I'm building an all-in-one mini-aloha and this was step one for me to scratch my itch! You could also checkout the Moss arm for a simpler build with limited 3D printing.

closed 2025-10-08 by pkooij