huggingface/lerobot #425: failed to read position in dynamixel.py

y-melamed, 2024-09-09:
### System Info

```Shell
- `lerobot` version: 0.1.0
- Platform: Linux-5.15.0-119-generic-x86_64-with-glibc2.31
- Python version: 3.11.8
- Huggingface_hub version: 0.24.6
- Dataset version: 2.21.0
- Numpy version: 1.26.4
- PyTorch version (GPU?): 2.4.0+cu121 (False)
- Cuda version: 12010
- Using GPU in script?: <fill in>
```

### Information

- [X] One of the scripts in the examples/ folder of LeRobot
- [ ] My own task or dataset (give details below)

### Reproduction

During teleportation or data recording we have come to read leader arm position error very often.
 "Read failed due to communication error on port". 

### Expected behavior

(all the code snippet are from common/robot_devices/motors/dynamixel.py)

When reading position of dynamixel motors (for the single arm koch robot in our case) we have come to this error a lot:

`        if comm != COMM_SUCCESS:
            raise ConnectionError(
                f"Read failed due to communication error on port {self.port} for group_key {group_key}: "
                f"{self.packet_handler.getTxRxResult(comm)}"
            )`

When we start to debug this we understood it related to problem in the positional data we read from our leader arm. 
for some reason it went out from the -270, 270 range very often and we got this exception:

`            if not np.all((values > -270) & (values < 270)):
                raise ValueError(
                    f"Wrong motor position range detected. "
                    f"Expected to be in [-270, +270] but in [{values.min()}, {values.max()}]. "
                    "This might be due to a cable connection issue creating an artificial 360 degrees jump in motor values. "
                    "You need to recalibrate by running: `python lerobot/scripts/control_robot.py calibrate`"
                )
`
We had to re calibrate too often and it was impossible to record data. 

We solve this by adding the following filter before:

`
        def adjust_values(arr):
            while np.any((arr >= 270) | (arr <= -270)):
                arr[arr >= 270] -= 360
                arr[arr <= -270] += 360
            return arr
` 

and run the code like this: 

`        if data_name in CALIBRATION_REQUIRED and self.calibration is not None:
            values = self.apply_calibration(values, motor_names)
            # print("vals after caliberation:", values)
            values = adjust_values(values)
`

also we also increase the value of NUM_READ_RETRY to 100. 

I'm not sure this need to be the default, but in our case it solved the issue:)

y-melamed, 2024-09-10:
Just saw that you implemented a solution in the current version, kudos!

Cadene, 2024-09-10:
Thanks @y-melamed ! Dont hesitate to re-open if you have any issue with calibration.