huggingface/lerobot #432: some questions about real world env

NNsauce, 2024-09-12:
### System Info

```Shell
all software cfg match author's project
```

### Information

- [ ] One of the scripts in the examples/ folder of LeRobot
- [X] My own task or dataset (give details below)

### Reproduction

I am planning to control my own robot left-arm. I've almost figure out all the parts if lerobot-dataset, then I want to make my own dataset respect to the aloha_sim_transfer_cube_human rather than "korch ALOHA teleop hardware system".
my questions are:
1) Must I keep such a high fps like 50 when collect data from camera and arm actions?
2) actions comes from human control on the arm, and state comes from reading operation, but how should I set the time gap between action and state?

### Expected behavior

answers from anyone

Cadene, 2024-09-12:
1. No need to keep high fps. We like fps=30 because it's quite smooth and doesnt put too much pressure on your computer ressources.
2. 
> "how should I set the time gap between action and state?"
Not sure I understand. As soon as we read from leader arm we send to follower.

NNsauce, 2024-09-14:
@Cadene thanks so much man!  
I am still not clear about question 2, are video and state recorded in a same thread func like this:
_def func():
     while True:
           read_frame()
           read_state()_
and after both are finished, then make time alignment?

NNsauce, 2024-09-14:
@Cadene thanks so much man!  On
For quetion 2 , I am still not clear about the time alignment of video and state, are they recorded in a same thread func like this?
def func():
    while True:
          read_frame()
          read_state()