huggingface/lerobot #435: Open-X datasets

nikonikolov, 2024-09-12:
Thanks for the great work! I am interested in converting more of the open-x datasets to `LeRobotDataset`.
- I was wondering if there was any particular reason the entire open-x wasn't added already, e.g. some difficulties you encountered with some specific datasets?
- Do you have any tips where I should be extra careful when converting from RLDS to `LeRobotDataset` or it's generally as easy as calling the conversion script?

Cadene, 2024-09-24:
cc @michel-aractingi for visibility.

nikonikolov, 2024-09-25:
Also Forgot to mention, some datasets have flipped image channels (BGR instead of RGB). Following https://github.com/kpertsch/rlds_dataset_mod/blob/main/prepare_open_x.sh:
    `berkeley_autolab_ur5`: flip_wrist_image_channels
    `stanford_hydra_dataset_converted_externally_to_rlds`: flip_wrist_image_channels,flip_image_channels
    `utaustin_mutex`: flip_wrist_image_channels,flip_image_channels
    `berkeley_fanuc_manipulation`: flip_wrist_image_channels,flip_image_channels

michel-aractingi, 2024-10-31:
Hey @nikonikolov,

Thanks alot for sharing your notes and experience its a huge help! 
I am doing a small refactoring of the Open-X datasets scripts and will simplify `transforms.py`. You're right I first got to the inspiration by how OpenVLA handles the Open-X datasets, but I will add all the missing raw data to the datasets. 

Part of the refactoring I think we should have a general rlds format and open-X to be special case of that format. So probably the format name might change to `rlds.openx.dataset_name`.

IrvingF7, 2024-12-30:
Sorry for jumping into this and don't mean to rush anybody. Just curious about the progress on this and if there is any ETA. Thank you guys for the great work.

nikonikolov, 2025-01-02:
Hey @michel-aractingi. Thanks for your help and work! Another point I didn't mention is that the current format of lerobot doesn't accommodate train-test dataset split. Some of the OXE datasets have explicit test partitions. I guess this is a lerobot-wide dataset format thing rather than specific to OXE. For reference, the way I handled is by making `meta`, `train`, `test` folders under the dataset root directory and a single `LeRobotDataset` python object corresponds to a single partition (either test or train).

closed 2025-10-08 by pkooij