huggingface/lerobot #436: mp4 format for images

nikonikolov, 2024-09-12:
I am quite interested in using `LeRobotDataset` for training. I am interested to get more context on the decision to store images in `.mp4` so I am aware of the implications this might have:
- Did you by chance study if the video compression has any negative effects on the image quality in terms of model performance (or any studies you based your decision on)
- I am looking to do large scale training. Do you have any data to compare if the speed for random dataset access is negatively impacted with mp4 compared to something like hdf5

Cadene, 2024-09-12:
We compared png frames versus mp4 video compressed on Pusht and Aloha environments in simulation. We didnt notice lower success rate. You could reproduce this result as we currently support both images and video datasets. For instance:
- https://huggingface.co/datasets/lerobot/pusht_image
- https://huggingface.co/datasets/lerobot/pusht
We might create a blog post about it at some point.

> "I see atm lerobot supports storing images either in .mp4 or .pt, but not in arrow or parquet format as many other HF datasets do."
As of now, we use parquet to store images: https://huggingface.co/datasets/lerobot/pusht_image/tree/main/data
We use parquet to store all other data (expect video that are in mp4 to be downloaded and streamed easily)

> Any ideas how pytorch would compare to arrow / parquet when using datasets of 100s of millions of examples?
Our current data format use parquet to store the data on hub, then arrow once it is downloaded in the cache (through HF dataset), and HF dataset load arrow data as pytorch tensors. It's fast enough for now. We are still iterating on the format to make it simpler and faster ; and scallable

richardrl, 2025-03-24:
Hi @Cadene , I am also interested in large scale training, perhaps with this library.
I am using about 20TB of data on an NFS network drive.
These options, were too slow for me:
1. Loading JPEGs from the filesystem
2. Decoding MP4s per batch

What worked was using FFCV, which processes the entire dataset into a binary file, shards it and preloads samples when starting each epoch.

I am not too familiar with these arrow /parquet etc. libraries. Do they have speed comparable to webdataset and FFCV? 

For me, the difference is about 20x or more, so it really matters to use a sharded storage format with pre-decoded images.

closed 2025-10-23 by github-actions[bot]