huggingface/lerobot #456: The instruction of using DynamixelWizard2 to update firmware in Readme will cause firmware update failure.

cometyang, 2024-09-30:
### System Info

```Shell
lerobot's low cost hardware configuration: Waveshare Serial Bus Servo Driver Board will brick the servo when update the firmware of servo
```

### Information

- [X] One of the scripts in the examples/ folder of LeRobot
- [ ] My own task or dataset (give details below)

### Reproduction

https://forum.robotis.com/t/firmware-transfer-failed-on-new-motor/5819

### Expected behavior

The readme (https://github.com/huggingface/lerobot/blob/main/examples/7_get_started_with_real_robot.md)
should not suggest updating firmware in https://github.com/jess-moss/koch-v1-1, the servo board will brick the servo.

sergesteban, 2024-09-30:
same happens here. updating the xl430-w250's firmwares caused that i cant see it anymore.
any solution with buying the U2D2?

cometyang, 2024-09-30:
i fix it by using Linux version wizard2.0.

sergesteban, 2024-10-01:
> i fix it by using Linux version wizard2.0.
> […](#)

I just tried and i cant read them either. can i ask u if u did a special config? connected straight to the servo? or anything else?

thanks in advance

chadd1783, 2024-10-24:
@sergesteban I fixed it by using by the U2D2 and doing a firmware recovery with the wizard.