huggingface/lerobot #458: Unknown encoder 'libsvtav1'

lxsy-xcy, 2024-10-02:
Hello! I find your work truly breathtaking. 
However, I've encountered an issue while converting my dataset to the LeRobot format using FFmpeg; 
it reports an "Unknown encoder 'libsvtav1'." 

Could you please advise me on how to resolve this? 
Alternatively, is there another encoder I can use? Would switching to a different encoder affect the policy training? 
Thank you!

lxsy-xcy, 2024-10-04:
install with conda-forge by running `conda install -c conda-forge ffmpeg` (it should be compiled with `libsvtav1`),

closed 2024-10-04 by lxsy-xcy

HANQAQ, 2025-02-10:
Hi, when I use the new version of Lerobot's SO-100 and run the following command:
`python lerobot/scripts/control_robot.py \
  --robot.type=so100 \
  --control.type=record \
  --control.fps=30 \
  --control.single_task="Grasp a lego block and put it in the bin." \
  --control.repo_id=${HF_USER}/so100_test \
  --control.tags='["so100","tutorial"]' \
  --control.warmup_time_s=5 \
  --control.episode_time_s=30 \
  --control.reset_time_s=30 \
  --control.num_episodes=2 \
  --control.push_to_hub=true`
the following errors occur:
`[vost#0:0 @ 0x13207240] Unknown encoder 'libsvtav1'
[vost#0:0 @ 0x13207240] Error selecting an encoder
Error opening output file /home/han/.cache/huggingface/lerobot/lyhhan/so100_test/videos/chunk-000/observation.images.laptop/episode_000000.mp4.
Error opening output files: Encoder not found`

and i use `conda install -c conda-forge ffmpeg`, but it dosen't work