huggingface/lerobot #474: tdmpc xarm example not working

fredh-leap, 2024-10-15:
### System Info

```Shell
lerobot version=0.1.0
os=Ubuntu 20.04.6 LTS
python version=3.10.15
numpy version=1.26.4
torch version=2.4.1+cu121
lerobots configuration='dataset_repo_id': 'lerobot/xarm_lift_medium',
 'device': 'cuda',
 'env': {'action_dim': 4,
         'episode_length': 200,
         'fps': '${fps}',
         'gym': {'obs_type': 'pixels_agent_pos',
                 'render_mode': 'rgb_array',
                 'visualization_height': 384,
                 'visualization_width': 384},
         'image_size': 84,
         'name': 'xarm',
         'state_dim': 4,
         'task': 'XarmLift-v0'},
 'eval': {'batch_size': 1, 'n_episodes': 1, 'use_async_envs': False},
 'fps': 15,
 'policy': {'advantage_scaling': 3.0,
            'cem_iterations': 6,
            'consistency_coeff': 20.0,
            'discount': 0.9,
            'elite_weighting_temperature': 0.5,
            'expectile_weight': 0.9,
            'gaussian_mean_momentum': 0.1,
            'horizon': 5,
            'image_encoder_hidden_dim': 32,
            'input_normalization_modes': None,
            'input_shapes': {'observation.image': [3, 84, 84],
                             'observation.state': ['${env.state_dim}']},
            'latent_dim': 50,
            'max_random_shift_ratio': 0.0476,
            'max_std': 2.0,
            'min_std': 0.05,
            'mlp_dim': 512,
            'n_action_repeats': 2,
            'n_action_steps': 1,
            'n_elites': 50,
            'n_gaussian_samples': 512,
            'n_pi_samples': 51,
            'name': 'tdmpc',
            'output_normalization_modes': {'action': 'min_max'},
            'output_shapes': {'action': ['${env.action_dim}']},
            'pi_coeff': 0.5,
            'pretrained_model_path': None,
            'q_ensemble_size': 5,
            'reward_coeff': 0.5,
            'state_encoder_hidden_dim': 256,
            'target_model_momentum': 0.995,
            'temporal_decay_coeff': 0.5,
            'uncertainty_regularizer_coeff': 1.0,
            'use_mpc': True,
            'value_coeff': 0.1},
 'resume': False,
 'seed': 1,
 'training': {'batch_size': 256,
              'delta_timestamps': {'action': '[i / ${fps} for i in '
                                             'range(${policy.horizon})]',
                                   'next.reward': '[i / ${fps} for i in '
                                                  'range(${policy.horizon})]',
                                   'observation.image': '[i / ${fps} for i in '
                                                        'range(${policy.horizon} '
                                                        '+ 1)]',
                                   'observation.state': '[i / ${fps} for i in '
                                                        'range(${policy.horizon} '
[... 166 lines cut ...]

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/mnt/batch/tasks/shared/LS_root/mounts/clusters/frednc41/code/Users/frederick.j.hoffman/lerobot/lerobot/scripts/train.py", line 652, in train_cli
    train(
  File "/mnt/batch/tasks/shared/LS_root/mounts/clusters/frednc41/code/Users/frederick.j.hoffman/lerobot/lerobot/scripts/train.py", line 326, in train
    eval_env = make_env(cfg)
  File "/mnt/batch/tasks/shared/LS_root/mounts/clusters/frednc41/code/Users/frederick.j.hoffman/lerobot/lerobot/common/envs/factory.py", line 51, in make_env
    env = env_cls(
  File "/anaconda/envs/lerobot/lib/python3.10/site-packages/gymnasium/vector/sync_vector_env.py", line 86, in __init__
    self.envs = [env_fn() for env_fn in env_fns]
  File "/anaconda/envs/lerobot/lib/python3.10/site-packages/gymnasium/vector/sync_vector_env.py", line 86, in <listcomp>
    self.envs = [env_fn() for env_fn in env_fns]
  File "/mnt/batch/tasks/shared/LS_root/mounts/clusters/frednc41/code/Users/frederick.j.hoffman/lerobot/lerobot/common/envs/factory.py", line 53, in <lambda>
    lambda: gym.make(gym_handle, disable_env_checker=True, **gym_kwgs)
  File "/anaconda/envs/lerobot/lib/python3.10/site-packages/gymnasium/envs/registration.py", line 752, in make
    raise type(e)(
TypeError: MujocoRenderer.render() got an unexpected keyword argument 'camera_name' was raised from the environment creator for gym_xarm/XarmLift-v0 with kwargs ({'obs_type': 'pixels_agent_pos', 'render_mode': 'rgb_array', 'visualization_width': 384, 'visualization_height': 384})

Set the environment variable HYDRA_FULL_ERROR=1 for a complete stack trace.

### Expected behavior

I was hoping to train a simple tdmpc policy to get started

suhyeok-jang, 2025-01-14:
same issue

TiantianLi28, 2025-06-30:
https://github.com/huggingface/gym-xarm/issues/9