huggingface/lerobot #477: Collecting human operated datasets in simulation

mmurray, 2024-10-17:
Hello,

Can you provide info on how human supervision was provided for the simulated datasets (e.g. `lerobot/aloha_sim_transfer_cube_human`)? I am starting to setup a similar MuJoCo gym environment for the Stretch (https://github.com/mmurray/gym-stretch) and I would like to collect/train on some human teleop data, but it seems like the current `control_robot.py` script and data collection examples are setup only for physical robots.

Thanks!

Cadene, 2024-10-18:
Hello @mmurray , we are currently working on adding `control_sim_robot.py` ;)
I let @michel-aractingi comment if he has time.

For `lerobot/aloha_sim_transfer_cube_human` you might find more info in the original aloha paper: https://arxiv.org/abs/2304.13705

Best

michel-aractingi, 2024-10-18:
Hello @mmurray 

You can find the `control_sim_robot.py` in this [branch](https://github.com/huggingface/lerobot/tree/user/michel-aractingi/2024-10-15-control-sim). Just keep in mind that this is not the final version and plenty of things will change especially with the new refactoring of `control_robot.py`. I have also only tested it with the mujoco environment in [gym_lowcostrobot](https://github.com/perezjln/gym-lowcostrobot/tree/main)

The usage is exactly the same as `control_robot.py`. You only need to define a sim config yaml file in `lerobot/configs/env`.
Here's an example of the one I am using with the `gym_lowcostrobot`. 
```
# @package _global_

fps: 50

env:
  name: lowcostrobot
  fps: ${fps}
  handle: PushCubeLoop-v0

  gym:
    render_mode: human
    max_episode_steps: 100000

calibration:
  axis_directions: [-1, -1, 1, -1, -1, -1]
  offsets: [0, -0.5, -0.5, 0, -0.5, 0] # factor of pi

eval:
  use_async_envs: false
```

mmurray, 2024-12-03:
Hello @michel-aractingi , thanks for this information. I looked at the `control_sim_robot.py` on the branch you shared and on your newer branch.  One thing I am still confused about is the robot configuration to use for simulation. When I don't use any config or if I try to use the koch.yaml config, the default manipulator robot tries to import the dynamixel sdk and I get `ModuleNotFoundError: No module named 'dynamixel_sdk'`.

I can install the dynamixel sdk locally, but I don't have any dynamixel devices so I still get errors `serial.serialutil.SerialException: [Errno 2] could not open port /dev/tty.usbmodem58760430441: [Errno 2] No such file or directory: '/dev/tty.usbmodem58760430441'`. 

Is there a robot configuration you use to collect data from a simulated robot on a machine without physical dynamixels available?

michel-aractingi, 2024-12-04:
Hi @mmurray , yes the simulation script is meant for teleoperating the simulated robot with a real leader arm. So far we tested on our manipulators. 

Could you provide me with more information on your setup? Do you have a link to the sim environment? With which device do you want to perform the teleoperation in sim?

junwuzhang, 2025-03-21:
Have a similar question. Can I run `control_sim_robot.py` to collect data in simulation, where I move the simulated robot either via a keyboard or a PS5 controller, but I don't actually have any real robot arm? Thanks! @michel-aractingi

jccalvojackson, 2025-05-07:
#1080

closed 2025-10-08 by jadechoghari