huggingface/lerobot #478: Dynamixel port finding not working for Windows.

diegoakel, 2024-10-20:
### System Info

```Shell
Function find_available_ports at lerobot/common/robot_devices/motors/dynamixel.py.
```

### Information

- [X] One of the scripts in the examples/ folder of LeRobot
- [ ] My own task or dataset (give details below)

### Reproduction

Running python lerobot/common/robot_devices/motors/dynamixel.py on windows. 

### Expected behavior

It's written only for Linux and Mac I guess. It works with this change:

```
def find_available_ports():
    return [port.device for port in serial.tools.list_ports.comports()]
```

eyast, 2024-11-13:
by any chance, are you able to proceed with the remaining of the tutorial? i.e., instantiate 
```leader_arm = DynamixelMotorsBus(
    port=COM8,
    motors={
        # name: (index, model)
        "shoulder_pan": (1, "xl330-m077"),
        "shoulder_lift": (2, "xl330-m077"),
        "elbow_flex": (3, "xl330-m077"),
        "wrist_flex": (4, "xl330-m077"),
        "wrist_roll": (5, "xl330-m077"),
        "gripper": (6, "xl330-m077"),
    },
)

Vacuame, 2024-11-13:
Thank you, it works! But I wonder if there's more code can't work in Windows——is it necessary to install Linux?

diegoakel, 2024-11-13:
i have been doing everything on Windows! For now the only problems i've seen is this one and the tutorial to connect the phone webcam with windows through droid cam.

pkooij, 2025-10-08:
Closing this as its resolved

closed 2025-10-08 by pkooij