huggingface/lerobot #483: Diffusion policy's RGB encoder shared by all cameras. Is this intended?

HiroIshida, 2024-10-22:
In the diffusion policy implementation, it seems that RGB encoder is shared by all camera images.
https://github.com/huggingface/lerobot/blob/c351e1fff92ed276ef668f188f645bdfdf42f6b3/lerobot/common/policies/diffusion/modeling_diffusion.py#L185

However, in the original paper, different image encoders are used for different cameras. 
[image]

@alexander-soare Is this design choice? (e.g. single encoder is better than multiple encoders?)

Chi, Cheng, et al. "Diffusion policy: Visuomotor policy learning via action diffusion." The International Journal of Robotics Research (2023): 02783649241273668.

alexander-soare, 2024-10-22:
@HiroIshida thanks for raising this. It was a choice for simple over complicated, mostly for speed of development. We're not against using multiple image encoders going forward.

HiroIshida, 2024-10-24:
@alexander-soare Thanks. I'm writing a modification. After checking on the real robot, I will make a PR.

closed 2024-10-30 by danaaubakirova