huggingface/lerobot #496: Eval with mps gives bad results

Xuzzo, 2024-11-02:
### System Info

```Shell
- `lerobot` version: 0.1.0
- Platform: macOS-14.7.1-arm64-arm-64bit
- Python version: 3.10.15
- Huggingface_hub version: 0.26.2
- Dataset version: 3.1.0
- Numpy version: 1.26.4
- PyTorch version (GPU?): 2.5.1 (False)
- Cuda version: N/A
- Using GPU in script?: <fill in>
```

### Information

- [ ] One of the scripts in the examples/ folder of LeRobot
- [ ] My own task or dataset (give details below)

### Reproduction

The diffusion_pusht agent from running 
```
python lerobot/scripts/eval.py \
    -p lerobot/diffusion_pusht \
    eval.n_episodes=10 \
    eval.batch_size=10
```
fails all eval episodes on mac.
After some digging, this is coming from pytorch/pytorch#139550

### Expected behavior

The same agent works with Cuda

alexis779, 2024-12-15:
Thanks for the hint.

@Xuzzo were you able to unblock training Le Robot on Apple GPU ?

It would awesome to make it work on my local MacBook Pro. I'd rather avoid purchasing a NVIDIA GPU or pay a cloud provider.

---

I'm using Apple M3 Max on MacOSX Sonoma 14.7.

I'm still seeing the issue. This is what I tried so far:

### Built torch & torchvision from source

I built `main` branch from today for `torch` and `torchvision` modules, within the conda environment.

```
% pip freeze | grep torch
-e git+https://github.com/pytorch/pytorch@5273d8fd2ab28157a183c5eb7633b9b20457e51d#egg=torch
-e git+https://github.com/pytorch/vision@f7b1cfa8f7e10e0c157da6e55dc6f0237397faec#egg=torchvision
```

### Torch

```
conda activate lerobot-dev
git clone -b main https://github.com/pytorch/pytorch
python setup.py develop
```

### Torchvision

I used this [trick](https://github.com/pytorch/pytorch/issues/103222#issuecomment-2543526756) to work around a compilation error.

```
conda activate lerobot-dev
git clone -b main https://github.com/pytorch/vision

brew install gcc

CC=/opt/homebrew/Cellar/gcc/14.2.0_1/bin/gcc-14 python setup.py develop
```

### Dependency cleanup

Remove `torch` and `torchvision` modules from `pyproject.toml` as they are now imported directly from the conda env.

```
% git diff pyproject.toml   
diff --git a/pyproject.toml b/pyproject.toml
index 59c2de8..4085caa 100644
--- a/pyproject.toml
+++ b/pyproject.toml
@@ -38,10 +38,8 @@ einops = ">=0.8.0"
 pymunk = ">=6.6.0"
 zarr = ">=2.17.0"
 numba = ">=0.59.0"
-torch = ">=2.2.1"
 opencv-python = ">=4.9.0"
 diffusers = ">=0.27.2"
-torchvision = ">=0.17.1"
 h5py = ">=3.10.0"
 huggingface-hub = {extras = ["hf-transfer", "cli"], version = ">=0.25.2"}
 gymnasium = "==0.29.1" # TODO(rcadene, aliberts): Make gym 1.0.0 work
```

Then run the cleanup of current module versions to fallback to the conda env versions.
```
poetry update
```

### Enable MPS device

```
diff --git a/examples/2_evaluate_pretrained_policy.py b/examples/2_evaluate_pretrained_policy.py
index b2fe1db..0f0bdc0 100644
--- a/examples/2_evaluate_pretrained_policy.py
+++ b/examples/2_evaluate_pretrained_policy.py
@@ -30,6 +30,9 @@ policy.eval()
 if torch.cuda.is_available():
     device = torch.device("cuda")
     print("GPU is available. Device set to:", device)
+elif torch.backends.mps.is_available():
+    device = torch.device("mps")
+    print("Running on Apple silicon chip. Device set to:", device)
 else:
     device = torch.device("cpu")
     print(f"GPU is not available. Device set to: {device}. Inference will be slower than on GPU.")
diff --git a/poetry.lock b/poetry.lock
```

### Run example script

```
% python examples/2_evaluate_pretrained_policy.py

Running on Apple silicon chip. Device set to: mps

Exception ignored from cffi callback <function ext_cpSpaceDebugDrawCircleImpl at 0x3f0eb4a60>:
Traceback (most recent call last):
  File "/opt/homebrew/Caskroom/miniconda/base/envs/lerobot-dev/lib/python3.10/site-packages/pymunk/_callbacks.py", line 103, in ext_cpSpaceDebugDrawCircleImpl
    options.draw_circle(
  File "/opt/homebrew/Caskroom/miniconda/base/envs/lerobot-dev/lib/python3.10/site-packages/gym_pusht/envs/pymunk_override.py", line 135, in draw_circle
    p = to_pygame(pos, self.surface)
  File "/opt/homebrew/Caskroom/miniconda/base/envs/lerobot-dev/lib/python3.10/site-packages/gym_pusht/envs/pymunk_override.py", line 231, in to_pygame
    return round(p[0]), round(p[1])
ValueError: cannot convert float NaN to integer
```

Xuzzo, 2024-12-15:
Your error is different to the one I had. I was able to run the script but the agent would fail all episodes. The workaround was to set "non_blocking=False".
Are you able to run the other eval scripts? I have the same setup (M3 Max with Sonoma 14.7) but had not problem installing torch. I am using torch==2.5.1 and torchvision==0.20.1

alexis779, 2024-12-15:
### 2_evaluate_pretrained_policy.py

Thanks! `non_blocking=False` fixed my issue. It mitigates the race condition ...

Following diff summarizes the overall change to run the script on Apple M3,

```
% git diff examples/2_evaluate_pretrained_policy.py
diff --git a/examples/2_evaluate_pretrained_policy.py b/examples/2_evaluate_pretrained_policy.py
index b2fe1db..af26445 100644
--- a/examples/2_evaluate_pretrained_policy.py
+++ b/examples/2_evaluate_pretrained_policy.py
@@ -30,6 +30,9 @@ policy.eval()
 if torch.cuda.is_available():
     device = torch.device("cuda")
     print("GPU is available. Device set to:", device)
+elif torch.backends.mps.is_available():
+    device = torch.device("mps")
+    print("Running on Apple silicon chip. Device set to:", device)
 else:
     device = torch.device("cpu")
     print(f"GPU is not available. Device set to: {device}. Inference will be slower than on GPU.")
@@ -73,8 +76,8 @@ while not done:
     image = image.permute(2, 0, 1)
 
     # Send data tensors from CPU to GPU
-    state = state.to(device, non_blocking=True)
-    image = image.to(device, non_blocking=True)
+    state = state.to(device, non_blocking=False)
+    image = image.to(device, non_blocking=False)
 
     # Add extra (empty) batch dimension, required to forward the policy
     state = state.unsqueeze(0)
```

### eval.py

I had no issue with the eval script using nightly build of torch / torchvision.

Did you pass `device=mps` when running the script?

Maybe the related pytorch [commit](https://github.com/pytorch/pytorch/commit/9bc9d4cdb4355a385a7d7959f07d04d1648d6904) fixed your issue?

```
% python lerobot/scripts/eval.py \
    -p lerobot/diffusion_pusht \
    eval.n_episodes=10 \
    eval.batch_size=10 \
    device=mps
Fetching 11 files: 100%|██████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████| 11/11 [00:00<00:00, 169001.26it/s]
INFO 2024-12-15 14:22:38 on/logger.py:39 Output dir: outputs/eval/2024-12-15/14-22-38_pusht_diffusion
INFO 2024-12-15 14:22:38 pts/eval.py:479 Making environment.
INFO 2024-12-15 14:22:39 pts/eval.py:482 Making policy.
WARNING 2024-12-15 14:22:39 s/factory.py:28 Hydra config is missing arguments: {'noise_scheduler_type', 'do_mask_loss_for_padding', 'use_separate_rgb_encoder_per_camera'}
Loading weights from local directory
Stepping through eval batches: 100%|█████████████████████████████████████████████████████████████████████████████████████████| 1/1 [01:24<00:00, 84.09s/it, running_success_rate=0.0%]
{'avg_sum_reward': 25.546307949020633, 'avg_max_reward': 0.10569534381657245, 'pc_success': 0.0, 'eval_s': 87.26703977584839, 'eval_ep_s': 8.726703977584839}                         
INFO 2024-12-15 14:24:08 pts/eval.py:509 End of eval
```

closed 2025-10-08 by pkooij