huggingface/lerobot #500: lerobot trying to delete tmp folder before video encoding is finished

1g0rrr, 2024-11-05:
### System Info

```Shell
Macbook AIR M1, python 3.10

- `lerobot` version: 0.1.0
- Platform: macOS-14.1-arm64-arm-64bit
- Python version: 3.10.14
- Huggingface_hub version: 0.25.2
- Dataset version: 2.21.0
- Numpy version: 1.26.4
- PyTorch version (GPU?): 2.2.2 (False)
- Cuda version: N/A
- Using GPU in script?: No
```

### Information

- [X] One of the scripts in the examples/ folder of LeRobot
- [ ] My own task or dataset (give details below)

### Reproduction

Install lerobot from scratch, record 5 episodes

### Expected behavior

I expected the tmp folder to be deleted only after the video finished encoding.

[image]

CarolinePascal, 2025-04-14:
Hi @1g0rrr !

Thanks for raising this issue. It seems to me that the problem you encountered was linked to a previous implementation of the `record` method. Have you tried to reproduce the error using the current `main` branch ?
Feel free to provide the error output if the problem persists !

Best,

Caroline.

1g0rrr, 2025-04-14:
Hi!
In new version it’ solved. Can be closed
Thanks fir fixing!

closed 2025-04-14 by 1g0rrr