huggingface/lerobot #502: Low accuracy for diffusion policy+aloha env+sim_transfer_cude_human dataset

Kimho666, 2024-11-08:
I'm trying to use diffusion model and aloha env to train on sim_transfer_cude_human dataset. But after 60000 training step, the evaluation accuracy is only 2%-6%. Idont know why? If I load pre-trained act policy, the accuracy can reach 80%.

Kimho666, 2024-11-08:
# @package _global_

# Defaults for training for the PushT dataset as per https://github.com/real-stanford/diffusion_policy.
# Note: We do not track EMA model weights as we discovered it does not improve the results. See
#       https://github.com/huggingface/lerobot/pull/134 for more details.

seed: 100000
dataset_repo_id: lerobot/aloha_sim_transfer_cube_human

override_dataset_stats:
  # TODO(rcadene, alexander-soare): should we remove image stats as well? do we use a pretrained vision model?
  observation.images.top:
    mean: [[[0.5]], [[0.5]], [[0.5]]]  # (c,1,1)
    std: [[[0.5]], [[0.5]], [[0.5]]]  # (c,1,1)
  # TODO(rcadene, alexander-soare): we override state and action stats to use the same as the pretrained model
  # from the original codebase, but we should remove these and train our own pretrained model
  # observation.state:
  #   min: [13.456424, 32.938293]
  #   max: [496.14618, 510.9579]
  # action:
  #   min: [12.0, 25.0]
  #   max: [511.0, 511.0]

training:
  offline_steps: 200000
  online_steps: 0
  eval_freq: 25000
  save_freq: 25000
  save_checkpoint: true

  batch_size: 64
  grad_clip_norm: 10
  lr: 1.0e-4
  lr_scheduler: cosine
  lr_warmup_steps: 500
  adam_betas: [0.95, 0.999]
  adam_eps: 1.0e-8
  adam_weight_decay: 1.0e-6
  online_steps_between_rollouts: 1

  delta_timestamps:
    observation.images.top: "[i / ${fps} for i in range(1 - ${policy.n_obs_steps}, 1)]"
    observation.state: "[i / ${fps} for i in range(1 - ${policy.n_obs_steps}, 1)]"
    action: "[i / ${fps} for i in range(1 - ${policy.n_obs_steps}, 1 - ${policy.n_obs_steps} + ${policy.horizon})]"

  # The original implementation doesn't sample frames for the last 7 steps,
  # which avoids excessive padding and leads to improved training results.
  drop_n_last_frames: 7  # ${policy.horizon} - ${policy.n_action_steps} - ${policy.n_obs_steps} + 1

eval:
  n_episodes: 50
  batch_size: 50

policy:
  name: diffusion

  # Input / output structure.
  n_obs_steps: 2
  horizon: 16
  n_action_steps: 8

  input_shapes:
    # TODO(rcadene, alexander-soare): add variables for height and width from the dataset/env?
    observation.images.top: [3, 480, 640]
    observation.state: ["${env.state_dim}"]
  output_shapes:
    action: ["${env.action_dim}"]

  # Normalization / Unnormalization
  input_normalization_modes:
    observation.images.top: mean_std
    observation.state: min_max
  output_normalization_modes:
    action: min_max

  # Architecture / modeling.
  # Vision backbone.
  vision_backbone: resnet18
  crop_shape: [84, 84]
  crop_is_random: True
  pretrained_backbone_weights: null
  use_group_norm: True
  spatial_softmax_num_keypoints: 32
  # Unet.
  down_dims: [512, 1024, 2048]
  kernel_size: 5
  n_groups: 8
  diffusion_step_embed_dim: 128
  use_film_scale_modulation: True
  # Noise scheduler.
  noise_scheduler_type: DDPM
  num_train_timesteps: 100
  beta_schedule: squaredcos_cap_v2
  beta_start: 0.0001
  beta_end: 0.02
  prediction_type: epsilon # epsilon / sample
  clip_sample: True
  clip_sample_range: 1.0

  # Inference
  num_inference_steps: null  # if not provided, defaults to `num_train_timesteps`

  # Loss computation
  do_mask_loss_for_padding: false

Kimho666, 2024-11-08:
This is my policy config. 
[diffusion4aloha.txt](https://github.com/user-attachments/files/17671776/diffusion4aloha.txt)

Kimho666, 2024-11-08:
[image]

CallMeFrozenBanana, 2024-12-04:
I adapted the codes to train diffusion policy on the aloha_sim_transfer_cube_scripted dataset and get the same results as you.
The EE always tries to clip a wrong position.
Just like below:
[image]

CallMeFrozenBanana, 2024-12-04:
> ![image](https://private-user-images.githubusercontent.com/137678908/384203965-427b4288-499f-41e3-ab12-766348e48700.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MzMzMDE0MjMsIm5iZiI6MTczMzMwMTEyMywicGF0aCI6Ii8xMzc2Nzg5MDgvMzg0MjAzOTY1LTQyN2I0Mjg4LTQ5OWYtNDFlMy1hYjEyLTc2NjM0OGU0ODcwMC5wbmc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQxMjA0JTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MTIwNFQwODMyMDNaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT05MzVkNGEyOWU4Y2ZiMDQxYWExNmUyOWM2OTAyYmQwZTkxZTY2YTZiZWUyMWQwM2U1YTdkNGU3ODdhY2ViZGUyJlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCJ9.OYgflE9gZ7qIf5nPqNXu9Lxt1ktks0_fbjrqO1h5Rzc)

I find that set a larger chunk size can slightly increase the success rate:
[image]

CallMeFrozenBanana, 2024-12-05:
The crop_shape you used is : [84, 84] which isn't appropriate to the 480 x 640 inputs. I have tried it but nothing works. Now I suspect that the problem is caused by the SpatialSoftmax layer.

CallMeFrozenBanana, 2024-12-06:
> The crop_shape you used is : [84, 84] which isn't appropriate to the 480 x 640 inputs. I have tried it but nothing works. Now I suspect that the problem is caused by the SpatialSoftmax layer.

Besides, change the cropping method to centered cropping, and you will find that everything works properly.
[image]

linCC111, 2024-12-06:
@CallMeFrozenBanana What is the configuration of your diffusion policy?

Taylor-Liu, 2024-12-08:
> Besides, change the cropping method to centered cropping, and you will find that everything works properly. [...]

Hi, what's your policy config and possible code modifications for this result?

CallMeFrozenBanana, 2024-12-09:
> Hi, what's your policy config and possible code modifications for this result?

Hi, you only need to change these two configs as follows:
  crop_shape: [440, 560]
  crop_is_random: False

Here is my config file ⬇️
```
# @package _global_

# Defaults for training for the PushT dataset as per https://github.com/real-stanford/diffusion_policy.
# Note: We do not track EMA model weights as we discovered it does not improve the results. See
#       https://github.com/huggingface/lerobot/pull/134 for more details.

seed: 100000
dataset_repo_id: lerobot/aloha_sim_transfer_cube_scripted
root: /data/dataset/HuggingFace-Download-Accelerator/hf_hub
 
override_dataset_stats:
  # TODO(rcadene, alexander-soare): should we remove image stats as well? do we use a pretrained vision model?
  observation.images.top:
    mean: [[[0.485]], [[0.456]], [[0.406]]]  # (c,1,1)
    std: [[[0.229]], [[0.224]], [[0.225]]]  # (c,1,1)
  # TODO(rcadene, alexander-soare): we override state and action stats to use the same as the pretrained model
  # from the original codebase, but we should remove these and train our own pretrained model
  # observation.state:
  #   min: [13.456424, 32.938293]
  #   max: [496.14618, 510.9579]
  # action:
  #   min: [12.0, 25.0]
  #   max: [511.0, 511.0]

training:
  offline_steps: 200000
  online_steps: 0
  eval_freq: 20000
  save_freq: 20000
  save_checkpoint: true

  batch_size: 20
  grad_clip_norm: 10
  lr: 1.0e-4
  lr_scheduler: cosine
  lr_warmup_steps: 500
  adam_betas: [0.95, 0.999]
  adam_eps: 1.0e-8
  adam_weight_decay: 1.0e-6
  online_steps_between_rollouts: 1

  delta_timestamps:
    observation.images.top: "[i / ${fps} for i in range(1 - ${policy.n_obs_steps}, 1)]"
    observation.state: "[i / ${fps} for i in range(1 - ${policy.n_obs_steps}, 1)]"
    action: "[i / ${fps} for i in range(1 - ${policy.n_obs_steps}, 1 - ${policy.n_obs_steps} + ${policy.horizon})]"

  # The original implementation doesn't sample frames for the last 7 steps,
  # which avoids excessive padding and leads to improved training results.
  drop_n_last_frames: 7  # ${policy.horizon} - ${policy.n_action_steps} - ${policy.n_obs_steps} + 1

eval:
  n_episodes: 25
  batch_size: 25

policy:
  name: diffusion

  # Input / output structure.
  n_obs_steps: 2
  horizon: 16
  n_action_steps: 8

  input_shapes:
    # TODO(rcadene, alexander-soare): add variables for height and width from the dataset/env?
    observation.images.top: [3, 480, 640]
    observation.state: ["${env.state_dim}"]
  output_shapes:
    action: ["${env.action_dim}"]

  # Normalization / Unnormalization
  input_normalization_modes:
    observation.images.top: mean_std 
    observation.state: min_max
  output_normalization_modes:
    action: min_max

  # Architecture / modeling.
  # Vision backbone.
  vision_backbone: resnet18
  crop_shape: [440, 560]
  crop_is_random: False
  pretrained_backbone_weights: null
  use_group_norm: True
  spatial_softmax_num_keypoints: 32
  # Unet.
  down_dims: [512, 1024, 2048]
  kernel_size: 5
  n_groups: 8
  diffusion_step_embed_dim: 128 
  use_film_scale_modulation: True
  # Noise scheduler.
  noise_scheduler_type: DDPM
  num_train_timesteps: 100
  beta_schedule: squaredcos_cap_v2
  beta_start: 0.0001
  beta_end: 0.02
  prediction_type: epsilon # epsilon / sample
  clip_sample: True
  clip_sample_range: 1.0
 
  # Inference
  num_inference_steps: null  # if not provided, defaults to `num_train_timesteps`

  # Loss computation
  do_mask_loss_for_padding: false
  use_l1_loss: False
```

Kimho666, 2024-12-28:
OK, thank u !

Kimho666, 2024-12-29:
[image]
[image]

I have modifiied the 'crop size' and 'random crop', but the success rate is also 4% (after 70000 steps). @CallMeFrozenBanana

Kimho666, 2024-12-30:
[image]
After using @CallMeFrozenBanana ‘s yaml, it always pick cude from the position as shown in the picture above. But the real cube is random.

Sui0206, 2025-02-11:
@Kimho666 Hello,
I have noticed that your 'num_train_timesteps' is only 10, i think this caused the bad performance of your policy, and I think you can just turn it to 100 to achieve a good result.

CallMeFrozenBanana, 2025-02-19:
@Kimho666 Sorry that I missed the message for so long. Have you solved this problem?

XJKunnn, 2025-02-22:
Hi, maybe you try to use a larger chunksize for aloha simulation environment by setting `n_action_steps` and `horizon` in config for diffusion policy:

```
horizon = 128
n_action_steps = 100
```

jiangranlv, 2025-04-28:
Hi, have anyone solved this problem now?

jiangranlv, 2025-04-29:
@CallMeFrozenBanana hi bro, i met the same issue and cannot reproduce your result with simply changing the crop size and crop center.

Kimho666, 2025-05-01:
> [@CallMeFrozenBanana](https://github.com/CallMeFrozenBanana) hi bro, i met the same issue and cannot reproduce your result with simply changing the crop size and crop center.

For pusht env, if you use act policy. The chunsize must be small such as 8. The accuracy will be improved. But for aloha env, I dont know.

Zelo-S, 2025-10-20:
It looks like the original paper mentioned using Transformer for Diffusion, specifically from miniGPT. Maybe that could be relevant for this kind of task?

ShengkaiWu, 2025-10-27:
Hi, have anyone solved this problem now?

Kimho666, 2025-11-28:
I am trying this config I will update the results later.

> CUDA_VISIBLE_DEVICES=0 lerobot-train \
>   --dataset.repo_id=lerobot/aloha_sim_transfer_cube_human \
>   --policy.type=diffusion \
>   --output_dir=outputs/train/2025_11_28/diffusion_aloha_transfer_cube_chunk128 \
>   --policy.device=cuda \
>   --job_name=diffusion_transfer_cube_chunk128 \
>   --wandb.enable=true \
>   --batch_size=32 \
>   --env.type=aloha \
>   --env.task=AlohaTransferCube-v0 \
>   --policy.push_to_hub=false \
>   --eval_freq=0 \
>   --steps=200000 \
>   --save_freq=50000 \
>   --policy.crop_shape=[440,560] \
>   --policy.crop_is_random=false \
>   --policy.horizon=128 \
>   --policy.n_action_steps=128

Kimho666, 2025-11-29:
set crop_is_random=false can improve success rate. 

[image]