huggingface/lerobot #509: `control_robot.py` does not work with Windows

dibbla, 2024-11-16:
### System Info

```Shell
- `lerobot` version: 0.1.0
- Platform: Windows-10-10.0.22631-SP0
- Python version: 3.10.15
- Huggingface_hub version: 0.26.2
- Dataset version: 3.1.0
- Numpy version: 1.26.3
- PyTorch version (GPU?): 2.4.0+cu124 (True)
- Cuda version: 12040
- Using GPU in script?: false
```

### Information

- [X] One of the scripts in the examples/ folder of LeRobot
- [ ] My own task or dataset (give details below)

### Reproduction

To reporduce, simply run the script on a Windows computer.

For example
`python .\lerobot\scripts\control_robot.py teleoperate`

This might be an issue with serial and DynamixelSDK, and the error message shows that 
`serial.serialutil.SerialException: could not open port '/dev/tty.usbmodem575E0032081'` which looks for a Linux-only directory.

#478 might be related, but it just fix the port finding, and does not applied to the problem in this issue.

Is there anyone can help or give some hints in solving this? I am happy to contribute if I can get more familiar with the whole process.

### Expected behavior

It should abllow me to teleoperate/control the robot.

aliberts, 2024-11-20:
Hi, thanks for reporting this. Have you tried the solution in the related issue you mentioned?
Changing the `find_available_ports` would allow you to find the correct port.
Apparently they used `port=COM8` in this other issue.

We don't test our code on Windows yet but if you can confirm the fix works, we can implement the change.

thefaftek-git, 2025-06-15:
It seems like the script DOES have this support now, it reported to me that it was on COM8 and didn't require any extra configurations to work.

TimPietrusky, 2025-06-15:
works also perfectly fine for me with my so100 on windows:

```bash
# calibrate
 python -m lerobot.calibrate --robot.type=so100_follower --robot.port=COM4 --robot.id=my_follower_arm

# teleoperate
python -m lerobot.teleoperate --robot.type=so100_follower --robot.port=COM4 --robot.id=my_follower_arm --teleop.type=so100_leader --teleop.port=COM3 --teleop.id=my_leader_arm
```

> [!NOTE]  
> replace the ports with the correct COM* for your computer

closed 2025-12-27 by github-actions[bot]