huggingface/lerobot #510: Do we have to compulsory use tarson robotics robots for this repo?

hemangjoshi37a, 2024-11-17:
Or any robot will work fine?

Also one more question.

Do we have to use depth camera or simple camera will work fine?

hemangjoshi37a, 2024-11-18:
Can you share any link to your work if possible? Like the robot you are using.

Sorry it is not tarson but some similar name that is the authentic robot that LeRobot uses for demos.

mydhui, 2024-11-18:
> Can you share any link to your work if possible? Like the robot you are using.
> 
> Sorry it is not tarson but some similar name that is the authentic robot that LeRobot uses for demos.

just follow the tutorial based on your budget and preference (I chose Moss)
 [moss](https://github.com/huggingface/lerobot/blob/main/examples/11_use_moss.md)
[koch](https://github.com/huggingface/lerobot/blob/main/examples/7_get_started_with_real_robot.md#4-train-a-policy-on-your-data)
[so100](https://github.com/huggingface/lerobot/blob/main/examples/10_use_so100.md)

they have hardware component buying lists and assemble videos

hemangjoshi37a, 2024-11-18:
ok thanks for response. but which is the easiest and cheapest ? I am kind of new in this so i want to start with small steps .

hemangjoshi37a, 2024-11-24:
Sorry mistook trossen for tarson robotics.

Cadene, 2024-11-25:
go for so100 ;)