huggingface/lerobot #515: ACT is working, but not Diffusion

Kacchan16, 2024-11-19:
### System Info

```Shell
Hello Team,

your work is so good, I am currently working on creating some nice policies with Lerobot repo, architecture and software. I tried ACT on my robot, it is working fine, able to execute the tasks what it learnt in the evaluation. 
I tried training Diffusion policy, multiple times with different params and also the default params, what you provided in the repo. I tried PushT in colab, its working but not in robot. Can you please explain why its not working, or should I change other things??
I forgot to mention, I used 3 cameras for data collection and training for Diffusion
Thank you
```

### Information

- [X] One of the scripts in the examples/ folder of LeRobot
- [ ] My own task or dataset (give details below)

### Reproduction

.

### Expected behavior

.

Cadene, 2024-11-19:
Thanks for reaching out @Kacchan16 

Could you provide more info? What are the behaviors that you observe with diffusion?

What configs did you try?

Kacchan16, 2024-11-19:
Thanks for quick reply, and sorry I forgot to mention the task, I am trying picking and placing the object first.

this is the config, I tried changing the resnet18 to resnet34, and offline_steps:200000 to 100000.
I reduced batch_size from 64 to 16, 8 bcoz of time and storage. and in starting I commented observation.state and action now I am training with the stats for my dataset.

Then Lerobot Diffusion wont work on real robot ?? ;(

"seed: 100000
dataset_repo_id: m2sher/pick_the_gear3

override_dataset_stats:
 observation.images.side:
   # stats from imagenet, since we use a pretrained vision model
   mean: [[[0.3885]],[[0.3795]],[[0.3613]]]  # (c,1,1)
   std: [[[0.3016]],[[0.2878]],[[0.2975]]]  # (c,1,1)
 observation.images.wrist_back:
   # stats from imagenet, since we use a pretrained vision model
   mean: [[[0.3964]],[[0.4092]],[[0.4035]]] # (c,1,1)
   std: [[[0.2899]],[[0.2737]],[[0.2644]]] # (c,1,1)
 observation.images.wrist:
   # stats from imagenet, since we use a pretrained vision model
   mean: [[[0.2892]],[[0.2904]],[[0.2771]]]  # (c,1,1)
   std: [[[0.2839]],[[0.2645]],[[0.2354]]]
 observation.state:
   min: [ 1.8578e-01, -6.8425e-02,  1.9700e-01, -2.6070e+01, -2.3693e+01, -1.0977e+01, -4.4053e-03]
   max: [ 0.5761,  0.0904,  0.5073, 19.9307, 10.0612, 62.7149,  1.0000]

 action:
   min: [ 1.7674e-01, -7.5004e-02,  2.0033e-01, -2.9240e+01, -3.1973e+01, -3.6199e+01, -4.4053e-03]
   max: [ 0.5810,  0.1336,  0.5132, 30.6663,  9.3582, 63.5602,  1.0000]

training:
 offline_steps: 200000
 #log_freq: 10
 online_steps: 0
 eval_freq: -1 
 save_freq: 25000
 save_checkpoint: true

 image_transforms:
    enable: true

 batch_size: 16
 grad_clip_norm: 10
 lr: 1.0e-4
 lr_backbone: 1.0e-5
 lr_scheduler: cosine
 lr_warmup_steps: 500
 adam_betas: [0.95, 0.999]
 adam_eps: 1.0e-8
 adam_weight_decay: 1.0e-6
 online_steps_between_rollouts: 1

 delta_timestamps:
   observation.images.side: "[i / ${fps} for i in range(1 - ${policy.n_obs_steps}, 1)]"
   observation.images.wrist: "[i / ${fps} for i in range(1 - ${policy.n_obs_steps}, 1)]"
   observation.images.wrist_back: "[i / ${fps} for i in range(1 - ${policy.n_obs_steps}, 1)]"
   observation.state: "[i / ${fps} for i in range(1 - ${policy.n_obs_steps}, 1)]"
   action: "[i / ${fps} for i in range(1 - ${policy.n_obs_steps}, 1 - ${policy.n_obs_steps} + ${policy.horizon})]"

 # The original implementation doesn't sample frames for the last 7 steps,
 # which avoids excessive padding and leads to improved training results.
 drop_n_last_frames: 7  # ${policy.horizon} - ${policy.n_action_steps} - ${policy.n_obs_steps} + 1

eval:
 n_episodes: 50
 batch_size: 50

policy:
 name: diffusion

 # Input / output structure.
 n_obs_steps: 2
 horizon: 16
 n_action_steps: 8

 input_shapes:
   # TODO(rcadene, alexander-soare): add variables for height and width from the dataset/env?
   observation.images.side: [3, 480, 640]
   observation.images.wrist: [3, 480, 640]
   observation.images.wrist_back: [3, 480, 640]
   observation.state: ["${env.state_dim}"]
 output_shapes:
   action: ["${env.action_dim}"]

 # Normalization / Unnormalization
 input_normalization_modes:
   observation.images.side: mean_std
   observation.images.wrist: mean_std
   observation.images.wrist_back: mean_std
   observation.state: min_max
 output_normalization_modes:
   action: min_max

 # Architecture / modeling.
 # Vision backbone.
 vision_backbone: resnet18
#  crop_shape: [84, 84]
# crop_shape: [96, 128]   
crop_shape: [480, 640]
 # crop_is_random: True 
 crop_is_random: False
 pretrained_backbone_weights: ResNet18_Weights.IMAGENET1K_V1 #null tomorrow
 use_group_norm: False
 spatial_softmax_num_keypoints: 32
 # Unet.
 down_dims: [512, 1024, 2048]
 kernel_size: 5
 n_groups: 8
 diffusion_step_embed_dim: 128
 use_film_scale_modulation: True
 # Noise scheduler.
 noise_scheduler_type: DDPM
 num_train_timesteps: 100
 beta_schedule: squaredcos_cap_v2
 beta_start: 0.0001
 beta_end: 0.02
 prediction_type: epsilon # epsilon / sample
 clip_sample: True
 clip_sample_range: 1.0

 # Inference
 num_inference_steps: 20  # if not provided, defaults to `num_train_timesteps`

 # Loss computation
 do_mask_loss_for_padding: false"

Kacchan16, 2024-11-21:
Hi all and @Cadene , 
I just wanted to ask one more question continous to previous one, Can you please just tell me this, Diffusion is not gonna work on real robot, what about VQ-BET and TDMPC ?

I thought of trying those policies on real robots. If you just tell me its gonna work or not, it will saves more time, rather than training with multiple parameters to make it work.

Thank you :)

Cadene, 2024-11-21:
Diffusion is expected to work on real robot ; but it requires some tuning imo.
Same for VQ-BET.

TDMPC is RL. It requires some rewards. We are working on making RL work well on real robots ;)

CallMeFrozenBanana, 2025-02-19:
Hi @Kacchan16 ~
Have you successfully applied the diffusion policy model in the real environment?
We encountered the same problem as yours.
The ACT performs well, but diffusion does not.

Kacchan16, 2025-04-16:
Hi @CallMeFrozenBanana 
No, but tried to make some necessary changes, for training the diffusion policy. But it didn't work, so tried with different policies.

MohamedAlaouiMhamdi, 2025-05-19:
Hi @Kacchan16,
Did diffusion or VQ-BET work for you? I have the same problem as you; I don't know why they don't work correctly.

closed 2025-11-30 by github-actions[bot]