huggingface/lerobot #517: Assertion Errors in Dataset Conversion

sangteak601, 2024-11-21:
Hi,

I am attempting to convert the dataset collected using the UMI gripper into a lerobot dataset format but keep encountering a failure at an assertion point.
The exact failure occurs [here](https://github.com/huggingface/lerobot/blob/main/lerobot/common/datasets/compute_stats.py#L140).
I am uncertain about the cause of this issue or potential solutions. For now, I have commented out the assertion section, which allows the dataset to be successfully converted.
Any advice you could provide would be greatly appreciated.

Command I run: 
```
python3 lerobot/scripts/push_dataset_to_hub.py --raw-dir /path/to/dataset --raw-format umi_zarr --repo-id myid/repo
```

Error:
```
Traceback (most recent call last):
  File "/lerobot/lerobot/scripts/push_dataset_to_hub.py", line 371, in <module>
    main()
  File "/lerobot/lerobot/scripts/push_dataset_to_hub.py", line 367, in main
    push_dataset_to_hub(**vars(args))
  File "/lerobot/lerobot/scripts/push_dataset_to_hub.py", line 230, in push_dataset_to_hub
    stats = compute_stats(lerobot_dataset, batch_size, num_workers)
  File "/lerobot/lerobot/common/datasets/compute_stats.py", line 140, in compute_stats
    assert torch.equal(first_batch_[key], first_batch[key])
AssertionError
```

Kacchan16, 2024-11-25:
Hi @sangteak601 
1. Can you tell me this, what is the dataset format you are converting ? and 
2. Are you collecting data with umi, while calculating data it is throwing error or different
Thanks

sangteak601, 2024-12-21:
@Kacchan16 Thanks for the reply.
It turns out that the dataset I collected had an issue, as the gripper opening part in the dataset was empty. After addressing this problem, it now works fine.

closed 2024-12-21 by sangteak601