huggingface/lerobot #525: Tran a RL agent (without initial dataset)

alexcbb, 2024-11-25:
Hi,

I'm currently working on trying to integrate the following environment in the repo : https://github.com/perezjln/gym-lowcostrobot
I would like to use it for learning a RL agent in sim and try it out on the real robot after.
However, the current training script requires to have a local or online pre-recorded dataset. Is there a way to avoid this and pass an option to not load a dataset ? 

Thank you in advance

alexcbb, 2024-11-27:
For now I've got a solution that is not optimal but seems to work : I generate a random dataset with some online interaction and then I launch the classical training script. If some people have better solutions let me know

michel-aractingi, 2024-11-27:
I'll that option thanks for the suggestions. Right now you can specify any dataset name and just set `training.offline_steps=0` and `training.online_sampling_ratio=1.0` in your command or policy config.

closed 2025-04-07 by imstevenpmwork