huggingface/lerobot #532: Waveshare ST3215 servo support

rabhishek100, 2024-11-28:
I am from India and I am able to only find the Waveshare ST3215 servo and Waveshare serial bus servo driver board locally here. Will the software work with that motor or do we need need to write a separate motor class for that to work?

rabhishek100, 2024-12-10:
The same code works!

closed 2024-12-10 by rabhishek100

Jeevesh2009, 2025-05-14:
Hey @rabhishek100 did you get any error like this while calibrating. I can run scan_motors.py and can position values for all 6 motors for 2 arms.

[image]

Jeevesh2009, 2025-05-23:
Hi @rabhishek100 , I ended up realizing that the frequency of reading the values from the motors is very low , try teleoperating fewer motors you wont get the error, I am able to teleoperate all 6 motors if i am adding some wait_length in the code. But since the freq is low and again i am inducing some delay , the motion is very shaky. Can anyone help? @kashif @danieldk @co42 I am using waveshare st3215 HS servos.

theharshithh-work, 2025-06-27:
hey @rabhishek100 did the calibration script work for you?

ananth-pal, 2026-02-16:
Got a runtime error as follows:

RuntimeError: Some Motors use different firmware versions:
{1: '3.10', 2: '3.10', 3: '3.9', 4: '3.10', 5: '3.10', 6: '3.10'}
Update their firmware first using Feetech's software. Visit https://www.feetechrc.com/software.