huggingface/lerobot #534: `--root` directory is not allowed to exit

apockill, 2024-11-30:
### System Info

```Shell
lerobot-1  | Traceback (most recent call last):
lerobot-1  |   File "/tmp/ee8739f1-22d7-410a-afe3-ab4eaf2289a7/lerobot/scripts/control_robot.py", line 555, in <module>
lerobot-1  |     record(robot, **kwargs)
lerobot-1  |   File "/tmp/ee8739f1-22d7-410a-afe3-ab4eaf2289a7/lerobot/common/robot_devices/utils.py", line 28, in wrapper
lerobot-1  |     raise e
lerobot-1  |   File "/tmp/ee8739f1-22d7-410a-afe3-ab4eaf2289a7/lerobot/common/robot_devices/utils.py", line 24, in wrapper
lerobot-1  |     return func(robot, *args, **kwargs)
lerobot-1  |   File "/tmp/ee8739f1-22d7-410a-afe3-ab4eaf2289a7/lerobot/scripts/control_robot.py", line 249, in record
lerobot-1  |     dataset = LeRobotDataset.create(
lerobot-1  |   File "/tmp/ee8739f1-22d7-410a-afe3-ab4eaf2289a7/lerobot/common/datasets/lerobot_dataset.py", line 925, in create
lerobot-1  |     obj.meta = LeRobotDatasetMetadata.create(
lerobot-1  |   File "/tmp/ee8739f1-22d7-410a-afe3-ab4eaf2289a7/lerobot/common/datasets/lerobot_dataset.py", line 291, in create
lerobot-1  |     obj.root.mkdir(parents=True, exist_ok=False)
lerobot-1  |   File "/usr/lib/python3.10/pathlib.py", line 1175, in mkdir
lerobot-1  |     self._accessor.mkdir(self, mode)
lerobot-1  | FileExistsError: [Errno 17] File exists: 'data'
```
```

### Information

- [ ] One of the scripts in the examples/ folder of LeRobot
- [ ] My own task or dataset (give details below)

### Reproduction

Run `control_robot.py record --root "data"` and it will fail if the `data` directory already exists. 

### Expected behavior

It should be okay to have many datasets under your root directory.

```
{root}/{HF_USER}/{dataset_name}
{root}/{HF_USER}/{dataset_name}
{root}/{HF_USER}/{dataset_name}
```

In the above example, only `dataset_name` should be forced to be unique.

apockill, 2024-11-30:
I'm a bit confused by lines around the codebase doing `Path(root) if root is not None else LEROBOT_HOME / repo_id`. 

Does this imply that `root` should already contain the full `root/user/dataset_name` within it? 

I'm happy to submit a PR once I know what the intended meaning of `root` is within the new 2.0 system!

aliberts, 2024-12-03:
Hi there,
As @Cadene mentioned in your PR, we changed the behavior of root to be directly the path where your dataset is stored.
in v1, you would do `--root=data` which would translate to
```python
root = Path("data") / repo_id
```
Now in v2, root directly points to your dataset with no `repo/id` intermediate directories.

That being said, we removed references to the `--root` to simplify thing in examples commands since by default, root will be at `~/.cache/huggingface/lerobot/repo/id` and specifying `root` should be necessary by default (but you still can if you want to).

aliberts, 2024-12-03:
I'm closing this since this is intended behavior but don't hesitate to re-open if you still think there's a problem ;)

closed 2024-12-03 by aliberts

apockill, 2024-12-03:
Awesome, thanks for the clarification!